motioneye-project/motioneyeos #418: Connection refused

anzabanza, 2016-07-25:
When I try to connect to the Pi's IP (port 80) I get a connection refused message (https://cloud.githubusercontent.com/assets/12784162/17113072/c998e80e-525d-11e6-8e58-9fb8f26a34ad.png). I tried with multiple devices and it doesn't work. Wireless and wired. Port 8081 does display an image but I'm not able to get to the control panel. It worked before but it suddenly stopped working. Please help!

ccrisan, 2016-07-25:
Hmm I'd need to see the logs. Please try to plug your SD card into a Linux machine and recover the logs from the /log directory on the data partition. Also, make sure you have enough disk space on your data partition.

anzabanza, 2016-07-27:
I don't have a sd card reader on my pc so I tried on my laptop but then I could't get kali linux to boot. When I tried too ftp to the pi it refused the connection too. So I just gave up and reinstalled motioneyeos and now the problem is solved. Thanks anyway.

closed 2016-07-27 by ccrisan