motioneye-project/motioneyeos #419: Port forwarding problem

blah4-5078, 2016-07-27:
ABSOLUTELY LOVE your program! Trying to get access of GUI away from home.
port forwarding= internal port 80 external port 20202 (picked random one)
BUT,,, port isn't open on multiple port checking sites, plus won't load on phone. please help! I don't think that i need DNS because i can memorize the IP address and port, but can do that if that is the problem.

http://wikisend.com/download/744262/boot.txt.log
http://wikisend.com/download/509620/dmesg.txt.log
http://wikisend.com/download/931150/messages.txt.log
http://wikisend.com/download/754358/motion.txt.log
http://wikisend.com/download/748186/motioneye.txt.log

stllunatic, 2016-07-27:
I was originally using the streaming option to view from my phone. After seeing your post I realized that the GUI can be viewed using the HTTP Port listed under the expert settings tab. The GUI is much more aesthetically pleasing from a mobile device.

My experience with getting this to work (RPi 3) after extensive troubleshooting is to be very specific with your router settings and port forwarding. Assigning a static IP address for my RPi seemed to help as well. TCP protocol was required for me. Be sure to double check all of your GUI: Network settings and Expert settings to make sure that they are compatible with your router's settings. Make sure you applied any settings made to either device and a reboot seems to help when changing things like this.

 Also, on a side note, while exploring, when I changed the HTTP Port in the GUI away from the default, the system rebooted but I was unable to access the GUI again even from the local IP address (I tried every number and port assigned, forwarded public and private). My router could see the RPi, I could SSH into it, and I could FTP into it. Only after I changed the config file back to default settings for the HTTP Port did the GUI return. The boot log showed "* Starting motioneye: failed" throughout.

Lesson learned: you can't change that port arbitrarily. I suppose each port has a designated purpose. 
[https://en.wikipedia.org/wiki/List_of_TCP_and_UDP_port_numbers](url)

More details on the (?) of HTTP Port section of the GUI would be helpful probably. 

```
>  shutting down motionEyeOS 20160705 ----
>  * Stopping motioneye: done
>  * Stopping smbd: done
>  * Stopping nmbd: done
>  * Stopping proftpd: done
>  * Stopping sshd: done
>  * Stopping crond: done
>  * Stopping date updater: done
>  * Stopping network: done
>  * Stopping wpa_supplicant: done
>  * Stopping eudev: done
>  * Stopping syslogd: done
> ---- booting motionEyeOS 20160705----
>  * Mounting filesystems: done
>  * Loading kernel modules: done
>  * Setting hostname: done
>  * Starting syslogd: done
>  * Configuring CPU frequency: done
>  * Starting watchdog: done
>  * Starting eudev: done
>  * Setting root password: done
>  * Starting wpa_supplicant: failed (brcmfmac)
>  * Configuring wireless network: no link
>  * Configuring wired network: no link
> ---- shutting down motionEyeOS 20160705 ----
>  * Stopping motioneye: failed
>  * Stopping smbd: failed
>  * Stopping nmbd: failed
>  * Stopping proftpd: failed
>  * Stopping sshd: failed
>  * Stopping crond: failed
>  * Stopping date updater: done
>  * Stopping network: done
>  * Stopping wpa_supplicant: done
>  * Stopping eudev: done
>  * Stopping syslogd: done
> ---- booting motionEyeOS 20160705----
>  * Mounting filesystems: done
>  * Loading kernel modules: done
>  * Setting hostname: done
>  * Starting syslogd: done
>  * Configuring CPU frequency: done
>  * Starting watchdog: done
>  * Starting eudev: done
>  * Setting root password: done
>  * Starting wpa_supplicant: done (brcmfmac)
>  * Configuring wireless network: ...
>  * Setting static gateway to ...: done
>  * Setting static DNS server to ....: done
>  * Setting current date using http: Tue Jul 26 21:30:08 CDT 2016
>  * Starting http date updater: done
>  * Starting crond: done
>  * Starting sshd: done
>  * Starting proftpd: done
>  * Setting smb admin password: done
>  * Starting smbd: done
>  * Starting nmbd: done
>  * Starting motioneye: failed
> 

```

blah4-5078, 2016-07-27:
Thank You SOO much for the quick response as I am going on vacation TODAY. (Probably shouldn't have waited to the last minute...) I will try it immediately!

blah4-5078, 2016-07-27:
Also, I changed the http port and was able to view it by adding : (whichever port it was changed to).

blah4-5078, 2016-07-27:
Still not connecting, port is still closed. Could it be my router?

blah4-5078, 2016-07-27:
[image]
Sorry for the multiple comments...

bre75, 2016-07-27:
Hi,

First i'd like to say MotionEyeOS is a wonderful solution.
Quite easy to setup.

I Just have a problem about dns forwarding.
I'm in France.
In France to access internet we use "box".
Which is kind like a modem/router and also managing wifi, landphone etc

To access MotionEyes front end from outside I create a No-IP address.
MyAddress.ddns.net

On my box there's an option to enable dynamic dns service.
I did it and no problem both are well linked.

Then i configure port redirection, as well as i create a rule on my firewall for the port i choose.

But still ... i can't access MotionEyeOS front end from outside my local network.

Any idea ?

Thanks a lot for any help.

Regards,

blah4-5078, 2016-07-29:
Okay I believe that the problem was that I have 2 routers one of them a modem also. I cascade port forwarded but I must've made a typo.

closed 2016-07-29 by blah4-5078