motioneye-project/motioneyeos #421: TOR question

bre75, 2016-07-28:
Hello,

I'd like my Raspberr PI/MotionEyeOS to use TOR.
Little bit the same way as they are doing there : [https://home-assistant.io/cookbook/tor_configuration/](url) 

In MotionEyeOS source code, here [https://github.com/ccrisan/motioneyeos/blob/master/package/Config.in](url), i can see a reference to TOR :  source "package/tor/Config.in"

Does it mean MotionEyeOS already use TOR network ?

Else how could i achieve this ?

Thanks for your help.

Regards,

ccrisan, 2016-07-28:
You can't use TOR with motionEyeOS. I recommend using Raspbian + motionEye instead if you need TOR.

closed 2016-07-28 by ccrisan

bre75, 2016-07-28:
Hi Calin,

Thanks for your answser.
Just for my curiosity what is the tor package in the MotionEyeOS source code ?

Instead of Raspbian + MotionEye, does a "Onion Pi" [https://learn.adafruit.com/onion-pi/overview](url) could do the job as well ?