motioneye-project/motioneyeos #422: MotioneyeOS as sever

ipglobal-chile, 2016-07-28:
Hello!, I have a doubt regarding the feature of MotioneyeOS. I would like to use this for create an architecture client/server only using MotioneyeOS. The clients will be Raspberry pi 2 with a camera attached and the server will be a Linux environment like ubuntu or arch (64 bit architecture). I have installed successfuly MotioneyeOS in the clients without issues, but I don't know how compile the motioneyeOS in the server side.

greetings from chile.

thanks!

ccrisan, 2016-07-28:
Install [motionEye](https://github.com/ccrisan/motioneye) on the server side.

ipglobal-chile, 2016-07-28:
Thanks ccrisan for your answer. I have followed the instruction for ubuntu server, (14.04 ver.). I can see the frontend through port :8765. I need to add a rtsp but I'm seeing a message error that is not support that protocol, in motioneyeOS (raspberry pi2 ) I can add them.

how I can give the feature in my motioneye server for rtsp protocol?

thanks again!

ccrisan, 2016-07-29:
How did you install motion?

ipglobal-chile, 2016-07-29:
Hi Ccrisan,

I have followed the below commands:
```

add-apt-repository -y ppa:kirillshkrogalev/ffmpeg-next
apt-get update
apt-get install python-pip python-dev curl libssl-dev libcurl4-openssl-dev libjpeg-dev
pip install motioneye
mkdir -p /etc/motioneye
cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
mkdir -p /var/lib/motioneye
cp /usr/local/share/motioneye/extra/motioneye.init-debian /etc/init.d/motioneye
chmod +x /etc/init.d/motioneye
update-rc.d -f motioneye defaults
/etc/init.d/motioneye start
pip install motioneye --upgrade

```
################################
`service motioneye restart`
#################################

ccrisan, 2016-07-29:
Where did you find these instructions? [These](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian) are the "official" ones for Raspbian on the Pi. Note that the motion daemon is compiled from Mr Dave's repo rather than using the stable version (which does not support RTSP, btw).

ipglobal-chile, 2016-07-29:
Ccrisan, I have resolve my issue upgrading the version of ubuntu from 14.04 to 15.10.

by the way, I would like to use my raspberry pi 2 as camera IP (I dont want to save files, or motion detection feature) and the server should have all the work of process the images. Currently the fps of raspberry pi 2 is like 2-15 fps with a V4L2 Camera (usb) _detected as /dev/v4l/by-id/usb-046d_HD_Pro_Webcam_C920_E293FD9F-video-index0._ Can I improve the fps only with one camera in raspberry pi 2 with 720p and 20-25 fps? what you think?

ccrisan, 2016-07-29:
If you want the best out of your PI as an IP cam I recommend using the official camera module as a Fast Network Camera.

ipglobal-chile, 2016-07-29:
Thanks Ccrisan. I will try to use one of them.

regards and thanks for your time.

closed 2016-08-05 by ipglobal-chile