motioneye-project/motioneyeos #425: UVC webcam was working with 20160410 release, no longer with 20160705

glaurent, 2016-07-30:
I was happily using a USB Creative Labs webcam which is UVC, it was working without problems with a previous release of motionEye but with the current release it no longer displays an image (I only get the barred camera icon).

I get this in dmesg :

[    3.280774] usb 1-1.3: New USB device found, idVendor=041e, idProduct=4058
[    3.280785] usb 1-1.3: New USB device strings: Mfr=1, Product=0, SerialNumber=0
[    3.280792] usb 1-1.3: Manufacturer: Creative Labs
[    3.308462] uvcvideo: Found UVC 1.00 device <unnamed> (041e:4058)
[    3.311126] input: UVC Camera (041e:4058) as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/input/input0
[    3.311273] usbcore: registered new interface driver uvcvideo
[    3.311280] USB Video Class driver (1.1.1)
[    3.354993] usbcore: registered new interface driver snd-usb-audio

any clues ? Should I downgrade ?

Thanks

ccrisan, 2016-07-30:
Hmm nope. Let's make it work as it should work with the latest version, unless there's a hardware fault. First off, did you try the latest Raspbian? motionEyeOS inherits the kernel used with Raspbian, and that's the only thing I can think of that might have changed between the two versions and could affect your USB camera.

glaurent, 2016-07-30:
Actually it's not working with the latest raspbian either, and I was surprised that it worked out of the box with motionEyeOS. So I guess that points to the raspbian kernel.

ccrisan, 2016-07-31:
Yeap, you should probably take this to the Raspbian forum (and keep us posted). In the meantime you should probably downgrade or use a different camera model.

glaurent, 2017-01-06:
Finally got around to post about it on the Raspbian forum : https://www.raspberrypi.org/forums/viewtopic.php?uid=187128&f=28&t=170382&start=0

I've also downgraded to 20160410.

ccrisan, 2017-01-06:
Thanks for keeping us posted. Let's hope the driver gets fixed soon. I'll make sure to update the kernel with each release.

bortek, 2017-06-01:
@glaurent whats the status of this?

glaurent, 2017-06-01:
Unchanged, as far as I know :-(.

StevenRaccoon, 2022-10-14:
They didnt even respond. Just locked the thread. What pricks.  Thatll give me some ammunition

starbasessd, 2022-10-14:
@StevenRaccoon How about using something a bit more current, instead of responding on a ticket that is 5+ years old?

StevenRaccoon, 2022-10-15:
Why?

starbasessd, 2022-10-15:
If you want help, vs just to complain, help can be found with using 20200606 and dev20201026. I don't know of anyone willing to help with 20160410. Many changes, and bug fixes were applied since then. I try to reproduce user issues to see if it is a bug or user issue, but, like most major software companies, I won't support, or even TRY to support something more than a few versions back. 
But, if all you want to do is 'file a complaint', consider it noted, and ignored.