motioneye-project/motioneyeos #427: Software Update 20160320 don't update

SuneBielefeldt, 2016-07-30:
I have a new PRiV3 not update to the next release.
The current version is 20160320, and I have tried to activate the pre-release, but without success.

Is there something I've missed, or should I just put a new version manually?

The unit has been unstable and there are numerous outages streaming.
It has otherwise been running fine for a few months at up to 10 fps

[image]

ccrisan, 2016-07-30:
I cannot seem to reproduce your problem. Can you please attach your motioneye.log file here?

SuneBielefeldt, 2016-08-01:
I think that WiFi connection was too weak. Moved device to my workshop, and here I could fine update it. The device is also stable here.

case Closed

closed 2016-08-01 by SuneBielefeldt