motioneye-project/motioneyeos #430: cant configure wifi - Starting wpa_supplicant: failed (brcmfmac)

aristosv, 2016-08-01:
I am running motioneyeos 20160705 on a raspberry pi 3. When connected to Ethernet everything is OK. When I configure it for WiFi I get the below during boot. Its a fresh installation. How can I fix this?

---- booting motionEyeOS 20160705----
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting watchdog: done
 * Starting eudev: done
 * Setting root password: done
 * Starting wpa_supplicant: failed (brcmfmac)
 * Configuring wireless network: no link
 * Configuring wired network: 192.168.7.23/24
 * Setting static gateway to 192.168.7.1: done
 * Setting static DNS server to 8.8.8.8: done
 * Setting current date using http: Mon Aug 1 13:23:30 EEST 2016
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting streameye: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.7.23/24
 # Default gateway is 192.168.7.1
 # DNS server address is 8.8.8.8

ccrisan, 2016-08-01:
Have you checked all the items from [here](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems)?

navigonic, 2016-11-06:
Same issue here, have checked all the items..

closed 2017-06-01 by bortek