motioneye-project/motioneyeos #431: Drive full / unable to log on

surfexplorer, 2016-08-01:
I've had the same issue a few times now where after a reboot I'm unable to log in via the web interface with a bad password error and from the console Im told there is not enough free space.  I'm using a 16GB SD card on my PI and a 450GB external drive for recording storage that is no way near full.  Each time this happens I reformat my sdcard and reset the system.  Any idea what I'm doing wrong?

Thanks

ccrisan, 2016-08-02:
Try identifying what's occupying that much space on your /data partition. It might be excessive logging (have you enabled Debugging in Expert Settings?), if you're sure media files go to the external drive.

surfexplorer, 2016-08-02:
I'll keep an eye on the data folder this time well in advance before the system fill up again leaving me locked out.  I suspect its a logging issue as well. This time I removed a faulty USB camera I had attached that may have been causing excessive logging.  Never enabled debugging before in expert settings and I have confirmed media files are going to the external drive.  Thanks

ccrisan, 2016-08-02:
What version is this by the way? Recent versions have logrotate support.

surfexplorer, 2016-08-02:
20160705

ccrisan, 2016-08-02:
That's odd. Please keep an eye on the log files and let's see if the logrotate mechanism is effective indeed.

fredericcollin, 2016-08-03:
I confirm the same login problems when the SD card is full with recordings (at least 99%). That has always happened accross the versions. The only way to solve the problem without re-installing the OS is to login with filezilla and erase some recordings with it (recover at least 5% of free space). Then it's possible to login again to MotioneyeOS the usual way.

closed 2017-06-01 by bortek