motioneye-project/motioneyeos #433: questions regarding motioneyeos web server

ivr588, 2016-08-02:
Hi, just a couple questions regarding the webserver (can't find the docs for it, though that might be my error) -

1. what's the webserver called? i.e what is it based on. What's the auth mechanism (htaccess? something else?)
2. does motioneyeos as shipped have any firewalling capability?

basically what I'm asking is, can it be exposed to the unfiltered internet safely and how resilient is the authentication mechanism.

thanks,

ccrisan, 2016-08-02:
1. Tornado; the authentication is a custom signature-based implementation.

2. Yes, motionEyeOS comes with the standard iptables Linux firewall command; you can use it to build your own firewall rules in [`/data/etc/userinit.sh`](https://github.com/ccrisan/motioneyeos/wiki/Tweaks#adding-custom-startup-commands).

I wouldn't recommend exposing it to the internet directly (at least not on port 80). I use an nginx-based reverse proxy that deals with HTTPS as well.

ivr588, 2016-08-02:
Many thanks for the rapid reply.

closed 2016-08-02 by ivr588

astbox, 2017-01-10:
Hello,

I am trying to put some iptables rules to the userinit.sh file to be added after reboot but for some reason my raspberry pi 3 with motioneyeos image (latest of time writing) gets stuck and I cannot access it after a reboot to execute the script.
The rules that I am trying to insert are the following

/usr/sbin/iptables -A INPUT -i lo -j ACCEPT
/usr/sbin/iptables -A INPUT -source 192.168.0.0/16 -j ACCEPT
/usr/sbin/iptables -P INPUT DROP

If I run locally the userinit.sh script the rules are being added normally without any problems.
The userinit.sh is executable and have #!/bin/bash at the start of the script in case that this matters.
My computer has a local ip of 192.168.1.x so normally I should have had access to the device.