motioneye-project/motioneyeos #435: Normal user only sees rotating circle, but no pic

maxl95, 2016-08-02:
Hello,

suddenly no "normal" user can view the images from the various Webcams, which I have configured. 
A "normal" user only sees the rotating circle but no pictures. 
When I log in as "admin" user, everything seems to be normal. 
I haven't changed anything. It occurred suddenly.

I attached the log files and two pictures for better understanding

Thanks

[boot.txt](https://github.com/ccrisan/motioneyeos/files/397171/boot.txt)
[dmesg.txt](https://github.com/ccrisan/motioneyeos/files/397169/dmesg.txt)
[messages.txt](https://github.com/ccrisan/motioneyeos/files/397173/messages.txt)
[motion.txt](https://github.com/ccrisan/motioneyeos/files/397170/motion.txt)
[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/397172/motioneye.txt)
[image]
[image]

ccrisan, 2016-08-14:
I'm sorry I cannot seem to reproduce this problem. Can you please attach your configuration files (you can recover them using the Backup function), after *removing sensitive data* from them?

ccrisan, 2016-08-20:
@maxl95 if you open the settings panel (after you have logged in with the normal surveillance user), what is the Preferences > Frame Rate Dimmer option set to?

closed 2017-06-01 by bortek