motioneye-project/motioneyeos #436: size of the page?

coME973, 2016-08-02:
Hello,

How can I adjust the size of videos to the size of the page?

As when configuring.

Thank you

ccrisan, 2016-08-02:
I'm not sure I understand what you want. Can you try to describe this on a screenshot (you could draw stuff on it)?

coME973, 2016-08-02:
I would like to see all the camera (4).
But when I put 2x2. We see nothing.
[image]
If I put it 4x1 are too small.
[image]
I would like the 2x2 grid adapts to the size of the window.

In the configuration menu is always good.
[image]
You understand?

coME973, 2016-08-02:
Anyway your system very nicely on a raspberry2.
I perform tests to see possibilities for professional projects.
Currently project but I hope to convince, if I manage to run the Action Buttons, particularly movement.

Can we get you a gift.

ccrisan, 2016-08-03:
Well it's quite hard to fit all frame sizes on all screens, preserving the *aspect ratio* of each frame (otherwise the image would look distorted). Ideally the "Fit Frames Vertically" could be a number (from 1 to 4) indicating how many frames to fit vertically within a window. In your case, it could be 2 (2x2). I'll take this into consideration for the next release.

coME973, 2016-08-07:
It is not possible to change the size of the video with the parameters in /etc/motioneyes/thread-1.conf ?
Extra Motion Options:
width 160
height 120

I have changed but nothing to do, only the size of the text changes.

[image]

ccrisan, 2016-12-04:
Commit https://github.com/ccrisan/motioneye/commit/bd90ce107622bd2b961765468613972afcd4266b should help out with this issue. The next release will include this fix.

closed 2016-12-04 by ccrisan