motioneye-project/motioneyeos #437: Stand-alone Video Recorder usage scanario

andrei-volkau, 2016-08-02:
Is It possible to make with help of motioneyeos a stand alone video recorder with the next features:
1) automatic restart of video recording after a power outage;
2) logging video to local usb HDD?

If It possible, say please few words about how to do It.

ccrisan, 2016-08-03:
1) Yes - it happens automatically; just enable movies and choose how you like movies to be recorded (motion-triggered or continuously).
2) Yes - just choose your USB storage device in File Storage.

closed 2016-08-03 by ccrisan

andrei-volkau, 2016-08-03:
Without a wired connection on-board computer have endless reboot loop.
[image]

I just want to preconfigure motioneyeos with help of motioneye. And after this, I want to use a single board computer with attached camera as a stand-alone dashcam without network connection.

ccrisan, 2016-08-04:
Hi,

You can't do that with motionEyeOS. However if you must have that, manually edit `/etc/init.d/S40network`, look for the line that says `"no network connection available"` and remove the following `reboot` command.

For more details on how to edit files on the system partition, see [Tweaks](https://github.com/ccrisan/motioneyeos/wiki/Tweaks).