motioneye-project/motioneyeos #438: Can't View Feed From Web Configurator

AMP648, 2016-08-02:
When I go to the :8081 I can view the stream but not when I go to the Configuration page on :80. I don't know why this is. The Pi is setup over Wi-Fi. Also I have authentication on for the Stream page on :8081. I don't know if that makes a difference. This is what I see:

In the Configuration Page(80):
[image]
In the Stream Page (8081)
[image]

Log Files:
[raspimjpeg.txt](https://github.com/ccrisan/motioneyeos/files/398222/raspimjpeg.txt)
[streameye.txt](https://github.com/ccrisan/motioneyeos/files/398223/streameye.txt)
[boot.txt](https://github.com/ccrisan/motioneyeos/files/398224/boot.txt)
[dmesg.txt](https://github.com/ccrisan/motioneyeos/files/398228/dmesg.txt)
[messages.txt](https://github.com/ccrisan/motioneyeos/files/398225/messages.txt)
[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/398226/motioneye.txt)
[motion.txt](https://github.com/ccrisan/motioneyeos/files/398227/motion.txt)

If someone could please help me figure out a way to view the stream from the configuration page that would be awesome. Thank you.

ccrisan, 2016-08-03:
I assume you have enabled Fast Network Camera. That's alright for streaming, but your browser might not be very happy with previewing mjpg streams. What browser are you on?

AMP648, 2016-08-03:
I was using Safari for Mac

ccrisan, 2016-08-04:
Have you tried another browser?

AMP648, 2016-08-05:
Yes, It works with Chrome. Thank you for your help. I love this software.

closed 2017-06-01 by bortek