motioneye-project/motioneyeos #439: New Camera added but motion is blurry

surfexplorer, 2016-08-03:
Just added a new ip camera to my motioneye (uokoo 720P IP Camera) but for some reason video playback and recording on motion eye is all blurry when capturing movement.  If I use the native web url the video is very clear and crisp and quality does not change when recording motion.  Any idea what I'm doing wrong?  Motioneye captures high quality video but just the movement is blurred.  My 10 year old USB camera works better and I just don't get it.

Thanks

ccrisan, 2016-08-04:
What version is this? What video codec and quality have you chosen?

surfexplorer, 2016-08-04:
version 20160705.  Not sure how to check video codec.  For recording quality I’ve tried 50% & 100% with same results.  Thanks

ccrisan, 2016-08-04:
Hmm can we see a screenshot of how blurry your video is? 20160705 won't let you choose the codec so that can't be a problem.

surfexplorer, 2016-08-11:
Very blurry as you can see here.  Still video and motion is crystal clear
when viewed using the built in camera web server.  Only blurry when used on
motioneye.  My USB camera on same system works just fine.  Thanks

[image: Inline image 1]

ccrisan, 2016-08-11:
Please attach a screenshot (here, on github, not via email).

surfexplorer, 2016-08-11:
[blurry.pdf](https://github.com/ccrisan/motioneyeos/files/413200/blurry.pdf)

ccrisan, 2016-08-11:
Well this is definitely not something related to motionEye but rather motion itself or the v4l2 camera driver (or the camera itself). I wouldn't know how to fix this.

closed 2016-08-11 by ccrisan