motioneye-project/motioneyeos #442: Motion Eye login

sunnymike, 2016-08-05:
I'm fairly computer illiterate. I got Motioneye installed on my raspberry pi and it is booted and asking me for a login....then a password (if I pick any login ). Where the heck do I find out what those 2 entries should be. No luck so far.

Help!!!!!!!!!

ipglobal-chile, 2016-08-05:
the default configuration below:
user: admin
password: (blank)

blah4-5078, 2016-08-05:
Same on command prompt

ans2016, 2016-12-29:
Hi, I am having the problem. 
I tried : user: admin, and  password: (blank), but it says Login is incorrect.
I also tried root and (blank), - No luck. 
I tried 2 different images - motioneyeos-raspberrypi3-20161027.img , and motioneyeos-raspberrypi3-20161212.img, and tried both logins, it did not work. Help will be much appreciated. Thanks.

ccrisan, 2016-12-29:
motionEye web login or ssh login?

ans2016, 2016-12-29:
In fact before the web login.
On my pi screen I get the following:

* Creating data partition: done
* Formatting  data partition: done
* Mounting data partition: done
......

Welcome to meye-80dcb569!
meye-80dcb569 login:

This is where  I am stuck.
I also tried another image - motioneyeos-raspberrypi3-20160410.img
This worked only for the first time, i.e with login:admin & password:(blank)
I managed to get the images from my webcam. I made a few changes
and the pi booted.
Then it is stuck on the login.

Thanks for helping.
Ans

ddauphin, 2016-12-29:
I've got the same issue here..
login: admin
password:

This combo is an incorrect login for me and I've never used this before.  I could not log in via the web; this was on my tv via hdmi and a usb keyboard.

ccrisan, 2016-12-30:
You're not supposed to log in directly using a monitor and keyboard (although it should work), but rather use the web interface.See https://github.com/ccrisan/motioneyeos/wiki/FAQ#i-am-only-getting-white-on-black-text-on-my-monitor-where-is-my-ui.

ans2016, 2016-12-30:
> only-getting-white-on-black-text-on-my-monitor-where-is-my-ui.

Thanks.
I tried, without raspbery pi screen and keyboard. The router does
not detect the IP of the pi. Looks like the pi is waiting for a login and
password to start.
Help plse.

Ans

ccrisan, 2016-12-31:
If the PI is waiting for a login, then it must have an IP address. Otherwise it would have rebooted. The IP address is displayed on the monitor right above the login prompt.

tejasasonavane, 2017-04-18:
I get the same thing. I enter admin as login and password (blank). After this i get 
[root@meye-xxxxxxxx ~]# 
Has anyone come across this? or has a solution to this?

cephasiii, 2017-04-18:
@tejasasonavane that is the command prompt for motioneyeos, it is the solution.

ccrisan, 2017-04-19:
Again, for reference: https://github.com/ccrisan/motioneyeos/wiki/FAQ#i-am-only-getting-white-on-black-text-on-my-monitor-where-is-my-ui.

closed 2017-04-19 by ccrisan

andrew-from-pretoria, 2017-04-22:
I had a similar problem: Console login did not work ('admin' with blank password). I had to reflash the SD card and start over. I think the problem was a corrupted OS because I had removed power a number of times without first shutting down the OS.  

TO confirm that you have a similar problem, watch the boot process on a connected monitor. For a correctly working SSD image, not much is displayed on the screen, at least much less than you would see with other OS such as Ubuntu. The 'broken'/ damaged image reported a number of 'files not found' and other errors during the boot process. In my case I was unable to login / access the device and this was because no IP address had been assigned to the board and so I had no address to point the browser to or the SSH terminal. 

The solution was to reimage the SSD card and reconfigure the program.

Pinewbie2017, 2017-09-05:
Hi everyone... (please) help, I am VERY new to this:
 i have tried many times, and have the same situation as (andrew) above ?
sorry but the posted answer makes no sense to me at all?
"@tejasasonavane that is the command prompt for motioneyeos, **it is the solution**.:"

NB: I have re-saved both NOOBs and MotioneyeOS for the Pi3 many times to a fresh micro sd card, SAME result?
 I just get the terminal prompt (trying to suggest is use the ip GUI) but have no idea how to achieve this ??

something is going wrong?
ALL help kindly appreciated, thank you.

MTijbout, 2017-10-15:
Hi all, Same here. After a couple of days, the set password is lost by the motionEye Pi. It keeps working (streaming and writing data to nas im my case) but no access to it anymore. Not even the User user. Re-flash is the only option. It happened to me in current and previous version a couple of times.

This is not a pebkac anymore to my opinion. I hope it will be fixed soon. Else I am forced to motion my eyes to alternatives...

My 2c.

ccrisan, 2017-10-16:
@MTijbout are you sure you're not dealing with #1116? See the last few comments for a possible way out.

MTijbout, 2017-10-18:
@ccrisan I've read it. I believe so. I believe I was changing the timezone too to get the right time in the picture stamps.

SixBytesUnder, 2017-11-12:
I had the same issue. I just updated to MotionEyeOS: 20171003 and was able to log in with my old password. I changed some other settings and applied them. After reboot, I was no longer able to log in with my old password. The password has been changed to `*****` - five asterisks.
Issue: saving any settings seems to be saving password to five asterisks.

BdgBill, 2017-11-20:
@SixBytesUnder - Just had the EXACT same issue. Updated a working system to 20171003 and got locked out. Old credentials would not work. I don't know how you figured out the asterisks thing but it saved me from reflashing. Thanks!

SixBytesUnder, 2017-11-20:
@BdgBill glad I was able to help others :) Years of experience as a web developer definitely helped me figure it out.  
Either way this is a harmless but incredibly annoying bug and I hope it will be fixed soon.

ccrisan, 2017-11-21:
@SixBytesUnder, @BdgBill the explanation as well as the solution with the asterisks `*****` was given in #1116, which as the original issue for this problem. The bug is in fact invalid as it is caused by an outdated, cached version of the javascript. Clearing the browser's cache fixes the problem. Future motionEye versions will include a version-specific query argument in the URL of the static resources, to prevent unfortunate situations like this one.

yusuf12061995, 2018-01-17:
Good morning, sir

I am yusuf from Indonesia, I want to ask for a motioneye.
how to change the appearance of raspberrypi title on motioneye? Can it ? or adding a sentence in the motioneye view ? can it ? please help me..... thankyou very much

merlinismycat, 2018-04-26:
hello everyone,
thank you very much, very good work ccrisan!
the project works from the first moment, and i am happy for this. :)

now I would like personalize the meye on my car type reverse cam and more.
you know someone that have started similary projects?

and one question, before of all, I woul like add multi video in usb like easycap chip  SAA7113 how can i do?

zraju, 2019-01-17:
When I log on using my mobile phone on my network; I can login in abut after that raspberry pi stops responding. I have to hard reboot pi?
But when I login using computer this does not happen. Any reason, please?

sergioaiello, 2019-02-25:
SOS!
I encountered the "invalid credential"@login issue three times in the last few hours. Every time I had to start from a scratch/formatted SD since I could not figure out the correct username/password combination (I did many many tries, also with their hashes, without success)

My conclusion is that MOTIONEYE inherits the password from the camera (or any other previous login).
In my last try I followed this procedure:
- as I added a new camera (the first of the system and a "network camera"), I was requested to insert camera's username and password, the browser asked to store them and I accepted => immediately motionEye registered that password in the control board. 
- i cleaned MotionEye "Admin Password" edit box and clicked on "Apply"
- MotionEye applied and asked me to login. I left the password field empty and it was ok
- I kept on configuring the video device/video streaming/motion detection/movies sections for the new camera
- Click on "Apply", MotionEye applied and asked for credentials: the password field was not empty and if blanked (as it should be) my login was rejected: "invalid credential". Now Login was possible with camera's password .... If I would have logged in to my email box then MotionEye would expect that password? I preferred not testing that when I inserted the email for email notification, but I can do if needed.
Then I removed the camera and the admin password to test the sequence again.
Click on "Add Camera" to get the popup window, but in background I saw that MotionEye filled up the password field again. Why??
After "Apply" the MotionEye asked for login and "password" field was already populated (with 5 dots as it did previously). I blanked and I got the "invalid credential" warning, so again MotionEye wrote not appropriately the password in the database.  

Here are the HW/SW info:
- motionEye Version: 0.39.3
- Motion Version: 4.0
- OS Version: Debian 9.8
- Linux orangepipc2 4.19.20-sunxi64 #5.75 SMP Fri Feb 8 10:29:25 CET 2019 aarch64 GNU/Linux (more detail here: http://ix.io/1BW3)
- Chrome Version 72.0.3626.109 (Official Build) (64 bit)
Please, let me know if additional info are needed.
Thank you for your support!

danialaris, 2020-03-04:
> Again, for reference: https://github.com/ccrisan/motioneyeos/wiki/FAQ#i-am-only-getting-white-on-black-text-on-my-monitor-where-is-my-ui.

i have changed the original ip address to a static ip address. Now i tried using the static ip address to open in web i can't. please help

danialaris, 2020-03-04:
> > Again, for reference: https://github.com/ccrisan/motioneyeos/wiki/FAQ#i-am-only-getting-white-on-black-text-on-my-monitor-where-is-my-ui.
> 
> i have changed the original ip address to a static ip address. Now i tried using the static ip address to open in web i can't. please help

@ccrisan help me please

starbasessd, 2020-03-04:
Are you using motiontEyeOS or motionEye? (motionEye runs on another OS like
Ubuntu, or Debian, motionEyeOS is stand alone)
MotionEyeOS WebGUI is   http://<IP-Address>
motionEye Web GUI is http://<IP-Address>:8765
If you connect a monitor to the PI running motionEyeOS it will tell you
what IP address it is using on the screen.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

merlinismycat, 2020-03-08:
Hi Cris,
I have solved everything.
I use motiontEyeOS in the PI3b raspberry
I changed the address and port to view the webcams via the web server in my Wago Plc.
thank you very much
and still a great job.

Ale

 

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Montanarx, 2021-08-28:
Aug 28, 2021

First MotionEyeOS  is wonderful software, I love it!  Thank you so much for all the work you put into this program.  

I had a power outage and was unable to log into MotionEyeOS.  I have it running on 5 different cameras.  Two would not let me log in. I reinstalled on the same card on a Raspberry Pi 0 with wifi dongle.  I could not upgrade the software because there was no value in the admin username field.  time zone was set to Africa.  Rebooted twice.  The second time the admin username field had a value.  Checked for updates and I have the latest version already installed.  

I have rebooted several times, everything seems to be working as expected.  

**For others having issues logging in:**
FYI my windows 10 machine keeps adding a space after admin (admin username).  If I type in admin I can get in.  If I go with the system and the added space I cannot log in.

starbasessd, 2022-06-29:
User:  admin
Password:  
If you haven't made any changes, there isn't a password

PGutti, 2023-01-22:
Hola! user/pass (as commented above) not accepted!

[image]

iubmadiwtu, 2023-01-23:
that's not motion eye that's home assistant
if you log in to motion eye it should look like this   
[image]
once you are in you can edit in this section 
[image]

you might need to set an admin & user password for it to work in HA

m3hm3tb4yr4m, 2023-07-10:
I have the same issue. When I flash the current version of motioneyeos on my SD card with writeimage.sh script or with balena etcher or with raspberry imager, I have no error. After I put my sdcard on my raspberry pi zero w. During the startup, I have this login and password to enter. I write "admin" for login and blank for password, I receive login and password incorrect. I note that the raspberry can't connect to my wireless network. I have wpa_supplicant.conf in boot partition of my sdcard. In otherwise, I have an error about "/data" partition. Could you help me please?