motioneye-project/motioneyeos #443: Video playback speed too fast

fredericcollin, 2016-08-05:
Hi ccrisan, I noticed on ver.20160801 that the playback speed of recorded videos is about twice too fast. Or did I set something wrong? 
I use two RPI2, one with the picam and the other with a webcam. Problem appears on both devices.
I tried to playback on other PCs and various player softwares and the speed problem was the same, so I think it's motioneyeos related.
Thank you for your time.

ccrisan, 2016-08-05:
There have been some changes regarding movies in 20160801. Can you please attach a screenshot of your movie settings as well as the video device settings (framerate & resolution)?

fredericcollin, 2016-08-05:
Sure! Here it is. Let me know if you need more info/data.
[image]

ccrisan, 2016-08-05:
Hmm can you please try reducing the video quality to, say, 75% - it might be too much for your PI and frames could be skipped (hence the fast playback).

fredericcollin, 2016-08-05:
OK I just tried and the video became extremely pixalated. Same speed problem. For example a 50 recording (as tracked per time stamp in the corner of the screen) takes only 30 seconds to play back, hence the fast pace.
I am confused...

ccrisan, 2016-08-05:
Could you try different qualities between 75% and 100%? I have changed the underlying parameter that controls the quality (supposedly the correct one, whereas before I was using the wrong one) but it seems this could pose problems.

fredericcollin, 2016-08-05:
I tried 85%, 90%, 95% and 98%. 
Up to 95% still pixelated. After 95% it gets better.
No matter what %, it always plays back too rapidly.

ccrisan, 2016-08-05:
Alright I'll take a look into this myself when I have a few free hours. Thanks for reporting the issue.

fredericcollin, 2016-08-05:
You are welcome. Thank you for making such a great O.S!

cuddylier, 2016-08-06:
I have the same issue too, hasn't bothered me too much though as it's much better than it used to be where it was just really pixelated and choppy but would be nice if it was just normal speed and watchable.

ccrisan, 2016-08-14:
I have identified the movie quality issue - it seems that the parameter that latest versions use has an exponential effect on the actual movie quality (as opposed to a linear one which I had hoped for). The next release will come with corresponding exponential transforms so that the 1 to 100% scale will actually correspond to the respective perceived quality (approximately, of course).

Regarding the fast playback, I myself am experiencing such an issue when using the mp4 (h264) codec.  It may be an issue with the way motion passes parameters to the ffmpeg library. Nevertheless, I have no idea how to fix this and it is definitely a motion thing, rather than a motionEye(OS) issue. Therefore I'll close the issue now and mark it as `motion`.

closed 2016-08-14 by ccrisan

jasaw, 2017-07-21:
@ccrisan Do you know what is the status of this bug? I can't find this bug in motion's issue list.

ccrisan, 2017-07-21:
@jasaw, regarding the fast playback issue I remember seeing similar problems when dealing with lower hardware (e.g. RPi1). I'm not sure if this has been tracked as a bug on motion but I remember seeing some discussions and commits to the way ffmpeg and motion deal with frame dropping. Things might have gotten better lately as a serious amount of motion commits were related to ffmpeg. I can't however say for sure if they fixed this particular issue. We'll see as soon as they release a new version incorporating all these changes.

ccrisan, 2017-07-21:
@jasaw would you mind if I added you as an admin for the motionEye(OS) repos?

jasaw, 2017-07-22:
@ccrisan There's a [commit](https://github.com/Motion-Project/motion/pull/275/commits/686b75a35fb2363c932b61deaf44ff35a524a329) in motion github repo that addresses pre-capture causing fast playback.

The version that I've tested already has this fix, but I'm still seeing fast playback, and I can't find anymore reference (no open issue, no commits) to fast playback. Probably movie PTS problem.

I'll have to play around with the source code to track down the problem.

I don't mind if you want to add me as admin. I'll try to help out when I can, which is highly dependent on my day job and family commitment.

cuddylier, 2017-07-22:
@jasaw Do you think it's worth me submitting an issue to motion repo about the poor FPS during recording/motion? I believe someone mentioned it was a motion issue before e.g. getting 5 - 10fps for 720p recording when you can get 30fps in Raspbian. CPU usage doesn't seem to be high at all during this to indicate a bottleneck somewhere.

ccrisan, 2017-07-22:
@jasaw I just added you as a collaborator. And thanks for helping out with so many issues around here.

ccrisan, 2017-07-22:
@cuddylier it might be worth opening an issue on the motion issue tracker if you can provide all required details (log files, configuration files, movie samples and a clear description of the problem). [This](https://github.com/Motion-Project/motion/wiki/motioneye) article may help with reporting motion issues when coming from motionEye(OS).

jasaw, 2017-07-24:
@cuddylier Depends on which hardware & software combination is giving you poor FPS.

The most common scenarios where poor FPS can happen:
- Recorded movie has low FPS. Possible causes:
  - Slow IO when writing to disk.
  - Motion software not using hardware accelerated encoder by default. I have enabled HW accel (for Raspberry Pis only) on my fork, but it may be worth getting motion developers to do something about this so it's in the motion mainstream.
- Fast Network Camera has low FPS. Possible causes:
  - not enough network bandwidth due to other traffic on the network, or poor wifi signal, or too much video data to send i.e. resolution & frame rate too high.
  - receiving end of the video stream unable keep up.
- Low FPS when watching from motioneye web UI (during motion/recording). Possible causes:
  - CPU usage too high on the device running motioneye.
  - not enough network bandwidth due to other traffic on the network, or poor wifi signal, or too much video data to send i.e. resolution & frame rate too high.

This is not an exhaustive list of possible set ups for motioneyeos.

It would be interesting to find out what kind of set up they have with Raspbian doing 30 fps. If you could provide more info, I'll look into it and maybe learn something from it.

cuddylier, 2017-07-24:
@jasaw Is the hardware deceleration built into your fork of motioneyeos itself? https://github.com/jasaw/motioneyeos I believe it was discussed before that the lack of hardware acceleration in motion may be playing a part.

I notice on your fork you mention relatively low quality and FPS specs to use as optimal. Have you tried 30 FPS 720p on a pi 3 yet to see if that works? Is the hardware acceleration enabled by default in your fork?

jasaw, 2017-07-26:
@cuddylier Yes, my fork has hardware acceleration enable, so it can achieve a higher frame rate than before, but that's not good enough because there are other bugs in motion, which I'm been busy fixing.

Regarding this fast playback issue, I've created a PR in motion project that fixes it.

cuddylier, 2017-07-30:
@jasaw Is there anyway to contact you about your fork? Seems you don't have an 'issues' section on it.

jasaw, 2017-07-31:
Fast playback issue has been fixed in motion master. Motion-Project/motion#431

@cuddylier I didn't know github disable issues by default. I've enabled it on my fork.

Komak57, 2020-04-01:
I've seen this issue for a while on Motion. VLC playback speed 0.02x plays at roughly the right speed, which means it's roughly 1FPS recording no matter what you set the recording speed to. It appears to be recording as if FFMPEG had `-r XX` set (which drops frames to achieve a framerate) rather than `-framerate XX`. Where this might be in the source is beyond my current understanding of Motion

jasaw, 2020-04-05:
@Komak57 Which motionEyeOS version are you running? Have you tried a different software to play back the video?

Komak57, 2020-04-05:
I've used VLC and FFPlay to watch the video, and FFProbe to inspect the
video data. I'll need to do recording tests with a clock to narrow down
exactly whats going on with the frames.

davebdb, 2020-07-06:
@Komak57 any update on what you did here?  I'm having similar issues when ffmpeg converts my High Predictive 4:4:4 video down to High IF the conversion using FFmpeg happens when the system has a high load.  (usually when all 4 cameras are going).  Not a problem with just 1 camera.  It also has problems if I go less than 100% and not do the conversion as then as the baseline profile is used.  So hope you can get some updated information.  I'll be posting here and with my thread as well with some tests I've done later tonight or this week, depending on what my time allows to put together my test results using ffprobe as well.