motioneye-project/motioneyeos #446: Unable to Boot Motioeyeos

vikasnagarajan, 2016-08-05:
Hi There... 

I have a Raspberry Pi 2 and have been using motionpieos for the past few months. It had been working fine. Recently I had to turn it off and did not use for a month or so. 

When I powered it back up (couple of days before), the device has not been booting up. Currently I see that the red light on Pi and Green light on Pi blinks every 5 seconds. Also, the light on Pi Cam turns on and remains on for 5 seconds. This process continues even if i leave my Pi connected for 10 mins. I have also connected my Pi to router but it does not detect my device (so no IP to login).

Any help on this appreciated. 

Thanks in advance.

sat5project, 2016-08-06:
connect the pi to a monitor using the HDMI output so you can read any error messages occurring at boot

vikasnagarajan, 2016-08-06:
I tried connecting to HDMI, I get no output (No Signal)...

scratchingclaw, 2016-08-09:
You could try to write a fresh disk image onto your Micro SD Card.

vikasnagarajan, 2016-08-09:
I tried that... But I get same result... Nothing changes... I even tried to write image on to a new SD card and tried it on another Pi as well...

scratchingclaw, 2016-08-09:
The second pi booted correctly?

sat5project, 2016-08-09:
should test the HDMI cable and the HDMI inputs on the display with separate devices

vikasnagarajan, 2016-08-09:
@scratchingclaw Even on the second Pi I had the same issue... 

@sat5project HDMI port seems to be working fine... I tried loading Raspbian on Pi and HDMI is working fine... 

Anyways, I installed Raspbian and loaded motioneye software on it to get the same running... 

Thanks for your inputs guys!!

closed 2016-08-09 by vikasnagarajan