motioneye-project/motioneyeos #447: Switched internet providers, different IP addresses, etc.

joemcglinchy, 2016-08-06:
Hello! I have switched internet providers due to a long distance move and as such all of my IP information has changed. My network SSID and password, however, have not, as I have ensured they are the exact same. Is there a way I can reconfigure MotionEyeOS to use my new network? Or will I need to re-image?

Thanks,
Joe

stllunatic, 2016-08-06:
I would try connecting your Pi via ethernet and seeing if you can update
the IP address on the web UI. If that doesn't work I would pull the micro
SD and update the config file manually on a PC. Also, it wouldn't hurt to
assign a static IP address to your Pi via MAC address so that your router
knows what to look for and matches the Pi's config settings.

A fresh image on the Pi wouldn't be a bad idea if you found it easy to
install the first time. If you have problems initially a few extra reboots
(power removal/install if no UI because of network issues) have helped me
in the past when changing settings that require a reboot.

sat5project, 2016-08-06:
the wireless network configuration file is `/data/etc/wpa_supplicant.conf `and can be edited after logging into a local console if you can connect a monitor to your device running motioneyeOS

closed 2017-06-01 by bortek