motioneye-project/motioneyeos #450: Multiple cameras mixing files together.

shevett, 2016-08-06:
I've set up a Raspberry Pi with a USB cam and the PiCam.  In the interface, they're visible as Camera3 and Pi-Cam.  However, no matter which one I look at and look at history, I only see the images from the PiCam, even though the subtitling says "Camera3".  Screenshots are attached here:

Here's the two cameras in the overview:
[image]

When I click on Camera3 and view history, I see:
[image]

I probably can manually changes the save path, but this should be handled in the interface.

ccrisan, 2016-08-06:
What are the values of the File Storage/Root Directory options of the two cameras?

shevett, 2016-08-06:
Here's what I'm seeing:
Storage device: Custom path
Root Directory: /data/output

I have not manually touched these since install.

I suspect the correct answer is to make that something like /data/output/picam and /data/output/camera3 - but when i add a new camera, it'll default to /data/output all the time, right?  Seems like a simple patch.

ccrisan, 2016-08-06:
Indeed, the `/data/output` is the default for all cams. The reason is that most users have one camera per motionEyeOS instance, motionEyeOS being intended for single-board computers which are usually not that powerful.

However it may be a good idea to separate the Root Directory for cameras from the start, automatically. I will take this into consideration.

shevett, 2016-08-06:
One last bit on this.  I manually edited the configs fro each camera specifying a new dir for each, but it looks like motioneyes needs to have the directories created by hand:

2016-08-06 15:46:16: [motioneye]    ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/output/picam'
2016-08-06 15:46:16: [motioneye]    ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/output/camera3'

Perhaps a hook on the web interface when you change the dir to say something like "This directory doesn't exist, do you want to create it?" would be helpful here.

ibedanny, 2016-08-11:
Does anyone have a recommendation for a cost effective ir camera I can add to my motioneye OS?

ibedanny, 2016-08-11:
WiFi IR camera.

ccrisan, 2016-08-14:
This issue has been (partially) fixed with https://github.com/ccrisan/motioneye/commit/7585d1858b78c186f7ee01f9710ae337e81e2f7c and https://github.com/ccrisan/motioneye/commit/b40825d03fae3fbe5fcb9830cedca7794d10fe8b.

Clem2Clone, 2016-08-15:
I'd like to see this options also. I will check your suggestions (partially fixed)

closed 2017-06-01 by bortek