motioneye-project/motioneyeos #451: Motion notification command error

sat5project, 2016-08-06:
Greetings,

I've written a simple python script to send an MQTT message from motioneyeOS on a raspberry pi3 to a remote server running an MQTT broker.  I **can** log into the pi3 and **successfully** run the script (send the message) manually from the command line but when I configure motioneyeOS to run the same command as a "motion notification", an error appears in motion.log.  

The issue was first present on the 20160705 build but I upgraded to the 20160801 build today and the issue is still occurring.  To be clear - I've _never_ successfully run this script with the 'motion notification' functionality.

Running a simple "Hello world" python script is successful and the STDOUT from the script does appear in motion.log as expected.
```
#!/usr/bin/python2.7
print "Hello, world"
```
  The failing MQTT script uses the PAHO libraries ( https://github.com/eclipse/paho.mqtt.python ) and produces the following error

```
Traceback (most recent call last):
  File "/data/paho/scripts/testpub.py", line 34, in <module>
    import paho.mqtt.client as mqtt
  File "/data/paho/usr/lib/python2.7/site-packages/paho/mqtt/client.py", line 53, in <module>
    if platform.system() == 'Windows':
  File "/usr/lib/python2.7/platform.py", line 1263, in system
  File "/usr/lib/python2.7/platform.py", line 1230, in uname
  File "/usr/lib/python2.7/platform.py", line 965, in _syscmd_uname
IOError: [Errno 10] No child processes
```

This error occurs and halts execution prior to any code being executed.  

The Paho library can be installed using the following steps: 

1. clone the Paho repo to another host where git is installed
2. scp/transfer the resulting directory to the motioneyeOS pi3 (preferably in /tmp)
3. installing the Paho library with the following command (run from within the transferred directory)

`python setup.py install  --root=/data/paho`

The library can then be imported from any script using the appropriate import path.   See testpub.py

I have chosen to first request assistance here rather than at the Paho github because the libraries work when run manually and only fail when attempting to use the "motion notification" functionality.  Any thoughts on why this might be failing would be most appreciated.  Relevant files uploaded.

[files.zip](https://github.com/ccrisan/motioneyeos/files/405137/files.zip)

Thanks,
James

sat5project, 2016-08-29:
I can't seem to locate where, in motioneyeos, the motion notification code is executed.  Can anyone point me in the right direction?

Thanks,
James

sat5project, 2016-09-02:
motion notification is executed here

https://github.com/ccrisan/motioneye/blob/master/motioneye/sendmail.py

but the amount of work required to recompile changes during troubleshooting was daunting enough that I found a shell script solution to sending MQTT messages which works fine:

https://github.com/raphaelcohn/bish-bosh

Thanks,
James

closed 2016-09-02 by sat5project

cap9qd, 2021-02-06:
I know this is horribly old but I found this when searching for the same error.  The key was in the sendmail.py you referenced:

With the signal override below allows me to import paho when executed from motion:

`import signal
#From MotionEye sendmail.py:
## the motion daemon overrides SIGCHLD,
## so we must restore it here,
## or otherwise media listing won't work
signal.signal(signal.SIGCHLD, signal.SIG_DFL)
import paho.mqtt.publish`

starbasessd, 2021-02-06:
@cap9qd better to start a new issue, rather than tag a comment on a closed issue, or reference the other referenced issue since it is still open. This ticket was started 4 1/2 years ago, and last mentioned 4 years ago...