motioneye-project/motioneyeos #452: Mobile Hot Spot

Just-Incase, 2016-08-06:
Hello All,

Just tried installing on rPi3 today.  Have HDMI plugged in and can see the process start and connect to wpa_supplicant file.  I have mobile hot spot turned on on phone, I see it connected to phone.  But I cannot get anything to happen with rPi IP adress in a web broser with :80, :8080 or :8081.

Any ideas?

My goal is to put in my car, I drive about 1,000 miles a week and wanted to share the link with my family.

Thank you!

stllunatic, 2016-08-06:
I've pondered this sort of project before too. It would be cool to make a
vehicle entertainment sytem out of a Pi as well. I researched and couldn't
find any specific info on this kind of thing. I think it may be difficult
to do this using a mobile hotspot because your hotspot isn't the same as a
traditional router and might not have ports. Maybe someone can explain how
a hotspot is different than a home ISP in more technical detail.

I would try installing Raspian on your Pi 3 and then install motioneye as
an application. This would maybe give you a little more freedom with
connecting to a mobile hotspot while on the move. From there you might be
able to upload pictures every 30s to a Google Drive that you can share with
your family.

sat5project, 2016-08-06:
The web browser you're opening would need to be running on a device that is also connected to your mobile hot spot.  It's _probable_ that a web browser running on your phone _should_ have connectivity with the pi but not guaranteed.  I would try from another device.

Just-Incase, 2016-08-06:
Thank you all!

I tried to veiw on the same device and from my home wifi.  Just no go.  I never thought about "ports".

Hope you all have a great day!

closed 2017-06-01 by bortek