motioneye-project/motioneyeos #453: File Storage Settings Not Showing Up

AMP648, 2016-08-08:
The file storage settings aren't showing up. The last section in my configuration page is video streaming.  Is this because of some of my settings?

[raspimjpeg.txt](https://github.com/ccrisan/motioneyeos/files/406222/raspimjpeg.txt)
[streameye.txt](https://github.com/ccrisan/motioneyeos/files/406225/streameye.txt)
[dmesg.txt](https://github.com/ccrisan/motioneyeos/files/406227/dmesg.txt)
[boot.txt](https://github.com/ccrisan/motioneyeos/files/406223/boot.txt)
[messages.txt](https://github.com/ccrisan/motioneyeos/files/406226/messages.txt)
[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/406224/motioneye.txt)
[motion.txt](https://github.com/ccrisan/motioneyeos/files/406228/motion.txt)

Sorry if this issue is due to ignorance on my part. I'm still new to this scene.

AMP648, 2016-08-08:
This might be just because I've enabled Fast Network Camera. I didn't know you couldn't save stuff while on Fast Network Mode.

AMP648, 2016-08-08:
Is there ever going to be a way to save files while using fast network camera?

ccrisan, 2016-08-08:
Nope. That will be impossible given the fact that FNC doesn't use motion as backend.

closed 2016-08-08 by ccrisan