motioneye-project/motioneyeos #455: Use without Internet

Just-Incase, 2016-08-09:
Hello,

I love this OS.  How can I use without a internet connection.  I want to use as a dashcam, but being it needs network to boot, I cannot.  How can I turn this off?

Thank you.

Just-Incase, 2016-08-09:
I would pay to have this option.  I really want it.

PiB4Dinner, 2016-08-09:
Ok you owe me Bitcoins! lol 

https://github.com/pageauc/pi-timolo
Look in the script of pi-timolo and compile the two or use pi-timolo only...
Time-lapse and motion recording. It works very well.
I have a 32GB card in mine and it is in a birdhouse; not for birds but for surveillance.  Bird hole covered by plexiglass and camera is inside the beast along with a DHT-22 Temp Humidity sensor. It has run non stop for a month and my card is not near full! 

-Wiz-of-OZ
Bitcoin, Cash, or Thanks is fine.

PiB4Dinner, 2016-08-09:
NO INTERNET NEEDED

Just-Incase, 2016-08-09:
@PiB4Dinner, how? Upon star up its looking for a network. How do you start the OS with a network? Thank you and have a great day.

Just-Incase, 2016-08-10:
@PiB4Dinner

I just watched the You Tub videos- awesome. I look forward to trying this. Looks intense. Will report back.

ccrisan, 2016-08-25:
Commit 16f67c02e056e2b7e9d6a4515ad580b3c959536c adds support for networkless configurations. Setting `link_watch="no"` in `/data/etc/watch.conf` will disable rebooting in case a network connection is not/no longer available.

closed 2016-08-25 by ccrisan