motioneye-project/motioneyeos #457: Working schedule not functioning

quantum8, 2016-08-12:
I have the working schedule set to only operate outside the hours of 06.30 - 17.30, but it is still recording everything that happens inside these hours.

Started happening after the 20160705 update.

[thread-1.conf.txt](https://github.com/ccrisan/motioneyeos/files/414635/thread-1.conf.txt)
[thread-2.conf.txt](https://github.com/ccrisan/motioneyeos/files/414636/thread-2.conf.txt)

ccrisan, 2016-08-12:
I will take a look.

ccrisan, 2016-08-20:
You have attached two camera config files, presumably corresponding to two cameras of yours. One of them has working schedule enabled, the other one doesn't. However they both share the same root directory and therefore media files go in the same directory. There's nothing wrong with that, but you might be seeing movies captured by one of them (the non-working schedule one) and think they were taken by the other one (the working schedule one). Isn't this the case?

quantum8, 2016-08-20:
I only have one camera connected at the moment, the one named IR. So everything that is recorded into that directory is only from that one camera, which is recording movements any time of the day. Is it possible that the scheduler only looks at the first camera settings and applies it to everything?

ccrisan, 2016-08-21:
There may be a mixup between cameras. You have two thread-*.conf files so you must have had two cameras some time. Try completely removing (not disabling) all cameras and re-add your desired camera from scratch. Then see if the problem persists.

quantum8, 2016-08-23:
Deleting the unused camera appears to have fixed this.

I'll have to test further with 2 cameras with different schedules to see if there is still a bug with this.

ccrisan, 2016-10-26:
eef55b4c24a02c7840f95faaf43c52276749eafb fixes this problem.

closed 2016-10-26 by ccrisan

surak, 2019-05-12:
This seems to be an issue after the latest upgrade somehow.

Erpiovra87, 2023-09-06:
@quantum8 can you help me how to find conf file for motioneye?

starbasessd, 2023-09-06:
@Erpiovra87 for which? motionEye or motionEyOS? Considering quantum8 last posted here in 2016...
The dev20201026 of motionEyeOS works fine.

Erpiovra87, 2023-09-06:
I've motioneye installed on raspi3 with home assistant. Current version: 0.19.1

starbasessd, 2023-09-06:
If you go to the webpage for motioneye (http://<ip_address>:8765 Then the Settings, General, it shows the version of motionEye there.

Erpiovra87, 2023-09-07:
0.43.0

starbasessd, 2023-09-07:
Open a new issue, it looks like Home Assistant installed the beta package for the Python3 only package, and it may be a bug within 0.43.0. 0.42.1 (motionEye installed from the github instructions and motionEyeOS dev20201026) used with Python2.7 motionEye and motionEyeOS has Working Schedule working just fine.

Erpiovra87, 2023-09-07:
Done. I've opened issue 2815. In the meantime i Will try to install 0.42.1. Do you have any link for istructions? Thanks in advance

starbasessd, 2023-09-07:
On the right hand side here: 
[Instructions for installing in various OSs](https://github.com/motioneye-project/motioneye/wiki/Installation)
Be aware, this may not be compatible with a HA install, depending on if using the HA OS (not going to work) install or if it is a docker or proxmox install  on one of the listed OSs.

Erpiovra87, 2023-09-07:
Thanks for suggest! I've installed raspi OS and motioneye 0.41.1 now it's working properly... you know of there Is any update about uploading files on Google drive?

starbasessd, 2023-09-07:
No idea, sorry. It's been almost 3 years since the last update, and the new devs have been 'less than rapid' since they took over...