motioneye-project/motioneyeos #458: notification in motion eye error

vm29, 2016-08-12:
hello?
we have an issue to set up the motion notification for our google account (gmail).
the error keep saying :

" Notification email failed:[Errno 97] Address family not supported by protocol! "

[image]

ccrisan, 2016-08-12:
Does that happen when you press "Test Email"? Have you entered your SMTP password correctly? (it doesn't look so from the screenshot).

vm29, 2016-08-14:
We already entered the smtp port but it still same. It cant be conduct
clearly as we enterted 587 for the port with gmail email..
should us have a try with another port or use another email account such as
yahoo?

>
> Does that happen when you press "Test Email"? Have you entered your SMTP
password correctly? (it doesn't look so from the screenshot).

vm29, 2016-08-15:
Oopps, my bad sir, I'll try to configure it again...if the problem still the same I will refer again with you..thank you

kgestrich, 2016-09-15:
Hello,
I have get the same error message: " Notification email failed:[Errno 97] Address family not supported by protocol! " when I try the email connection with "Test Mail"
[image]

Ich try to connect to a myname@gmail.com account. So far I have tried several variations to send out a notification mail - unfortunatly without any success...
- I changed the password to only contain small and capital letters and numbers. 
- the 2 step verification is turned off
- Less secure apps are allowed: https://www.google.com/settings/security/lesssecureapps
- signing in a new device https://accounts.google.com/DisplayUnlockCaptcha 
- I have no problem with the internet connection - I can ping the smtp.gmail.com server
- the password and username / email is working fine in the web-interface.

What should I change / try now? Have you got any idear or can I send you more information?

Thank you very much for your help in advance.
Greetings
Klaus (kgestrich)

ccrisan, 2016-09-15:
See if port 587 works

kgestrich, 2016-09-15:
:-D great it is working!
Thanks a lot!!! - and it was such a small stupid thing ...

closed 2016-09-15 by ccrisan

psycoppp, 2016-12-04:
Have the same problem and dont understand why.
I verify my adresse and pw, it's ok.
i used MotionPie (version 20150719) since a month, with the same mail, and it's work.
Yesterday, i install MotionEye, and it's dont work  :'(
The same Raspi, the same IP and the same mail.
Any idea ?
[image]