motioneye-project/motioneyeos #459: NoIR Camera not seeing IR light

NeiRodgers, 2016-08-12:
Hi. I am really enjoying MotionEye OS but am having issues with the Raspberry Pi NoIR camera which does not respond to or detect IR light. I have 2 separate sources, a security multi led spotlight (not sure what wavelength) and also the light from a Nest Cam (850nM). The Nest cam works with both but the NoIR cam doesn't like either.

Pi Hut are going to replace the camera but I am not sure that it is faulty. MotionEyeOS detected and installed the camera without me having to change anything. Do I need to do anything to configure the camera to work with IR ? Does anyone have this working OK with the same set up ?

TIA  Neil

ccrisan, 2016-08-12:
The NoIR camera "sees" IR light automatically, out-of-the-box and without any tweaking. It physically lacks an IR filter and it is naturally sensible to IR light. If it doesn't show anything but darkness in dark, IR-lit environments, most likely it is not a NoIR camera.

NeiRodgers, 2016-08-12:
Thanks ccrisan. I suspected as much though it looks to be the correct one. I have sent it back anyways for a replacement. The normal colours were not like a normal camera. Ill report back when the new camera arrives. Thanks for the super fast response ! Have a great weekend.

bortek, 2017-02-23:
@ccrisan this issue can be closed. I still have received my collaborator access for the project.

ccrisan, 2017-02-23:
@bortek my apologies, I forgot to add you to the collaborators. I hope you have now received the invitation.

closed 2017-02-23 by ccrisan