motioneye-project/motioneyeos #460: android and ios app ?

nextgen1979, 2016-08-13:
Hello , 

i'm searching atfer an android and ios app to view my motioneyeos camera 

do you have some idea ?

i have also one qnap for storage 

thanks for your help

stllunatic, 2016-08-13:
Hey, I have a Galaxy phone and I drop a Google chrome link on my home page to my MotioneyeOS and it works exactly like what I would expect from an app. I can access settings very easily as well.

nextgen1979, 2016-08-15:
i will try thanks

closed 2017-06-01 by bortek

ryanginnelly252, 2021-05-06:
Hi There, I'm looking for something similar for iOS. I know I can add the webpage to my home screen but, is there any apps recommended?

starbasessd, 2021-05-06:
There are no 'officially supported' IOS or Android apps. motionEye and motionEyeOS are written to support most, if not all web browsers. If you need to access from outside your home network, that involves setting up port forwarding on your home router, and you might get support here, but I don't support that, there are too many different routers, and too many different ways to set them up.

ryanginnelly252, 2021-05-06:
Okay.