motioneye-project/motioneyeos #461: hostname with dot not supported

auban, 2016-08-14:
Hello,
I would like enter a hostname with dot but it not supported by the frontend.
Is there a reason?

Thank you

Auban le Grelle

ccrisan, 2016-08-14:
What version do you use?

auban, 2016-08-15:
I have install the last release.
motioneyeos-raspberrypi3-20160801.img

[image]

ccrisan, 2016-08-15:
Ah, hostname changing is supported, but don't use special characters (remove the dot).

auban, 2016-08-15:
I would like use a domain name

ccrisan, 2016-08-15:
On a second thought, I believe there's nothing wrong with allowing dots in hostname. I'll make sure that the next version will allow dots in hostnames.

auban, 2016-08-16:
Great many thanks

kcranley1, 2016-08-16:
Continuing on from the Facebook conversation, I can assign a Surveillance Username of . (dot) with a blank Surveillance Password, but when I try to log in with those details, the rotating icons start, but just keep on rotating (ie it doesn't get any further).  My version is Current Version	20160801  What I really want to do is allow the video to be displayed on a web page without the need for viewers to log in.  I greatly appreciate your attention in this matter.

ccrisan, 2016-08-16:
Normally the correct way to have a passwordless surveillance system is by simply clearing out the surveillance password while leaving the surveillance username to whatever it is (e.g. "user").

However I can see that in your case it doesn't work that way. Does the system still ask for credentials or it simply goes to the camera frames view where the actual video streams never show up?

kcranley1, 2016-08-16:
It simply goes to the camera frames view where the actual video streams never show up

ccrisan, 2016-08-16:
Alright. Could you please try opening the UI webpage in Chrome and then open the Developer Tools (Ctrl + Shift + J). After that, a printscreen of the Developer Tools window could be revealing :)

kcranley1, 2016-08-16:
Is this what you mean?
[image]

ccrisan, 2016-08-16:
I found the problem. Somehow the Frame Rate Dimmer preference is set to 0%. This means that frames will never refresh.

ccrisan, 2016-08-16:
Also please remove the screenshot now (or at least blur the IP address) or otherwise unwanted users may visit your website.

kcranley1, 2016-08-16:
Thanks - have you fixed something?  It's just the same here.

ccrisan, 2016-08-16:
I haven't touched it. Just open the settings panel and increase the Frame Rate Dimmer (set it to 100%, preferrably). Then hit refresh and you should be fine.

kcranley1, 2016-08-16:
I did that, but the problem's just the same.

ccrisan, 2016-08-16:
I changed it myself now to 40% and it appeared to have been fixed.

kcranley1, 2016-08-16:
Yes that works.  Thank you so much for your considerable help with this amazing software.  Now can you tell me how I should be displaying the video without a security risk?

ccrisan, 2016-08-24:
Fixed with 88c76e9504e8bc0c05dca0fac65a9d991faf5a55.

closed 2016-08-24 by ccrisan