motioneye-project/motioneyeos #463: No sign of Action Buttons

mcastaneda83, 2016-08-17:
Hi,
I have been using 20160705 and 20160801 images on a Rpi Zero which is using a Raspi Cam V2.1, everything works great but I don't get to make the action buttons appear on top of the camera frame.
I have followed the steps described on the Action Buttons wiki article, the "light_on" example to be more specific.  On the other hand, my gpio wiring seems to be ok.
Any clue of what could I be doing wrong? 

Thank you and congralutations for this awesome project.

ccrisan, 2016-08-18:
Have you made them executable?

mcastaneda83, 2016-08-18:
Yes I forgot to mention that, I made "chmod +x" on the created bash script.

ccrisan, 2016-08-18:
Where did you place them and what are their names?

mcastaneda83, 2016-08-18:
I put them in /data/etc
They are called "light_on_cam1" and "light_off_cam2"

ccrisan, 2016-08-18:
Just rename them to "light_on_1" and "light_off_2". I assume you have two cameras; one of them will have a "light on" button and the other one a "light off"

mcastaneda83, 2016-08-18:
It works! Thank you so much.

closed 2016-08-18 by ccrisan