motioneye-project/motioneyeos #464: Run motion only when MAC not on LAN

snagglewest, 2016-08-18:
Ccrisan,

A good while back in the motioneye days I had asked about the possibility of incorporating a method to disable a camera when a particular mac address (cell phone) was present.  For example, when the wife or I come home the interior cameras would cut off and when we leave start back up.  It's just a bit creepy to record inside while we're home, not to mention it takes a lot of storage space.

My thought is a setting to input 3-5 mac addresses that motoioneyeos would check for and start motion when the mac is not on the network and stop motion when they show up.   It's well past my knowledge level but these guys seem to have a script figured out that looks like it would be fairly easy to incorporate.  Of course that's from my non-programmer prospective so I might be far off base.

Your thoughts?

[http://superuser.com/questions/493003/how-to-run-webcam-software-only-when-i-am-not-home-phone-is-not-on-the-lan](url)

joshhighet, 2016-08-21:
This sounds like a great idea and something that could really come in hand.

bortek, 2017-06-01:
Use a time schedule instead.

closed 2017-06-01 by bortek