motioneye-project/motioneyeos #465: Remote MotionEye Camera, request timed out...

SuneBielefeldt, 2016-08-18:
## Remote MotionEye Camera, request timed out...

When I try to connect a Remote Motioneye Camera, I meet often a "Request timed out" after about 30 sec.

I try again many times, finally succeeded, but I do not understand why this time out comes so quickly. 
Is it something that can be changed ? ?

In the first time with Motioneye, it gave rise to great frustration because I thought that something was done wrong.

[image]

ccrisan, 2016-08-19:
It's most likely a connectivity issue - something that has nothing to do with motionEye(OS).

SuneBielefeldt, 2016-08-20:
The mentioned Remote Motioneye cameras, sitting on an internet connection 20 km away. So there may well be many sources of error.

The next day I tried again, and here it was possible to set up 9 Remote Motioneye cameras.

Thanks for the fabulously good work you do to keep this fantastic project in the air.

closed 2016-08-20 by SuneBielefeldt