motioneye-project/motioneyeos #466: unable to open picture browser via UI

sroettgermann, 2016-08-19:
I'm unable to open the picture browser via the webinterface. The system seems to load, but after a while I get back to the start screen and nothing happens. 
Further I want to upload the still images to my Fritz!Nas or my Dropbox. But it seems not to work, too.

ccrisan, 2016-08-19:
You probably have too many media files. Please try cleaning up your media files folder, making sure that there are no more than a few hundred photos in that folder. On a Raspberry PI with a standard SD card, it's understandable to have such performance issues.

sroettgermann, 2016-08-19:
Thanks for the quick answer. That might be a reason. How do I cleanup the files properly?

ccrisan, 2016-08-19:
Using ftp, sftp or smb (network shares).

closed 2017-06-01 by bortek