motioneye-project/motioneyeos #468: motioneye and mypi

sunnymike, 2016-08-20:
Rather than use WebIOPi I would like to use MyPi as my GPIO interface. If I do not have motioneye embedded into Raspberry Pi, can I embed MyPi somehow into motioneye?

ccrisan, 2016-08-21:
Use motionEye on Raspbian then.

closed 2016-08-21 by ccrisan

sunnymike, 2016-08-21:
I have tried to put motionEye on Raspbian using this:

http://www.howtoembed.com/projects/raspberry-pi/95-motioneye-with-raspberry-pi#comments

as the method.

When I try to install this I get an error message:

sudo aptitude install python-tornado python-jinja2 python-imaging motion ffmpeg v4l-utils

The error message basically says it cannot find a suitable ffmpeg package then doesn't install any of the packages so motioneye will not work without this.

Michael Sundgaard
mgsundgaard@verizon.net

ccrisan, 2016-08-21:
How about the instructions described [here](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian)?

sunnymike, 2016-08-21:
I'll try it.
Thanks