motioneye-project/motioneyeos #471: Frame Change Threshold question: constant images being sent

Clem2Clone, 2016-08-22:
I have 2 USB cameras attached along with the pi camera, monitoring my home. The main one is a 120 degree wide angle unit that sends pics to Dropbox constantly. I have set frame change threshold to 20%, noise level 7%, light switch det 75%. The room gets a lot of light during the day. I am showing frame changes and I can see changes looking at the camera but there is no movement in the scene.

Clem2Clone, 2016-08-23:
The 2 cameras that are connected along with the basic RaspIon, send pictures continuously. I've tried changing settings but still get many pics sent seemingly without any motion. Can you advise?

andrewvictor, 2017-02-24:
After reading through comments, this seems like a good place to jump in.

I have a similar issue.  I was able to tune my Frame Change Threshold which, to my understanding, controls the threshold at which motion is triggered for a MINIMUM amount of pixel changes frame to frame.  The issue with a room that sees a lot of light, as Clem2Clone mentioned, is this creates a large frame to frame pixel change, which always triggers motion.  

1. Are we able to use the Light Switch Detection to control a MAXIMUM pixel threshold for motion detection?  Would you be able to explain this feature?  

2. Could this be a feature added for the next build?  Instead of a minimum change threshold, possibly a pixel change threshold range?  ie.  5000pixels, MIN 2% = 100, MAX 70% = 3500 ; motion triggered between 100-3500 pixels

Thanks!

P.S.  This software package is the real deal!  Extremely please with the work done here!

andrewvictor, 2017-02-24:
@ccrisan

ccrisan, 2017-02-24:
@andrewvictor motionEye doesn't have that much control over what these parameters do. It's merely a frontend for motion and as far as I know, there's no way you can instruct motion to detect changes between a minimum and a maximum limit. You could however ask this question on the motion issues page or their forum.

andrewvictor, 2017-02-24:
@ccrisan Thank you for your fast response.

Are you able to comment on what exactly the Light Switch Detection does?  Or should I direct this to motion?

ccrisan, 2017-02-24:
[this](http://htmlpreview.github.io/?https://github.com/Motion-Project/motion/blob/master/motion_guide.html#lightswitch) is the best explanation of what the lightswitch detection actually does. You could use it to somehow limit the amount of frame "changes" that are considered an event, but I'm not sure how that is implemented or that the value can actually be used to transform image size percents into pixels.

andrewvictor, 2017-02-24:
@ccrisan This explanation is perfect.

Thanks again.

closed 2017-06-01 by bortek

tigrrr, 2017-07-29:
@ccrisan is there any way to finetune the motion detection treshold ? the slider allows minimum change 0.1 % which is the change i'm trying to detect (cca. 200 pix on a megapixel camera) 

Maybe you can explain how to tweak the motioneye UI interface slider maximum Treshold value from 20% to something like 5 % ?
Thanx :)

ccrisan, 2017-07-30:
@tigrrr given my experience with motion and motionEye, I really don't think 0.1% pixels changed would make any difference when it comes to motion detection. But if you really want that, edit the threshold manually in your thread-1.conf and the UI won't (hopefully) change it anymore. You'll need to compute the number of pixels based on the percentage and the image dimensions.

Felixrising, 2018-01-26:
Sorry, bumping an old thread...  

motion uses number of changed pixels for "threshold" configuration value with it's default of 1500 pixels (a bit arbitrary, but previous recommendations where around "2000 for indoors and 1000-10000 for outdoors"). 

Motioneye is taking a percentage value and applying it to the total frame in this line: https://github.com/ccrisan/motioneye/blob/e35a6cf866ce3cf5b2a0887a7eb019828e55fca0/motioneye/config.py#L837 
Given the 1500 default is about 0.49% on a 640x480 (307200 pixel) image, and motioneye is using one decimal place, tuning on a VGA image is limited to about 271 pixel increments/decrements. With about 10000 being about 3.2% and 1000 being 0.3%*. I'll also note that when masking is being used, this becomes worse as potentially large portions of the image are being masked making the actual number of pixels likely to be changed even smaller. It seems that the current 0-20% slider limits the tuning a bit. 

*https://htmlpreview.github.io/?https://raw.githubusercontent.com/Motion-Project/motion/master/motion_config.html#threshold

Would it be possible to have an threshold pixel integer field in addition to the % slider? And as a moonshot feature request, would it be possible to have a pixel change histogram for tuning purposes? It could use a RRD to capture the pixel delta over time.

MoonCactus, 2019-02-17:
Sorry also bumping an old thread

I need it to be about 150px (less than 1%), as 80% of my frame is masked/useless for detection. The default 0-20% threshold of motioneye is not fine enough to set the threshold in this case.
I simply had to tweak the nicely done motioneye web ui so that the slider allows for more precision, namely by setting the range in `/usr/lib/python2.7/site-packages/motioneye/templates/main.html` directly on the device, e.g.:
```
<tr class="settings-item advanced-setting" min="0" max="5" snap="0" ticksnum="5" decimals="2" unit="%">                                                  
  <td class="settings-item-label"><span class="settings-item-label">Frame Change Threshold</span></td>
```
Of course I also had to bump decimals to 2 for such a small interval, and had to `mount -o remount,rw /dev/root /` in the first place. Note that this "fix" be overridden with any upgrade.

jasaw, 2019-02-18:
I also find the motion detection threshold slider a bit too coarse, so I ended up editing `/data/etc/thread-1.conf` by hand.