motioneye-project/motioneyeos #472: Delete oldest images when full

John12341234, 2016-08-22:
I'm sure there is a reason for why a "delete images that are older than x days" algorithm was selected over a "delete oldest day when free-space is less than 2%" algorithm, but I can't imagine how that reason could be justified. 

I'm sure it is obvious, but just in case it isn't, the current algorithm allows the SD card to fill up if the user doesn't correctly _guess_ how many days of images will fit on it. Not only that, but each day results in a different number of images that are stored, depending on the amount of activity. These two reasons make it an inferior algorithm to one that simply deletes the oldest day when a free-space threshold is reached. Not only that, but it also reduces the number of settings that need to be set by the user. 

My guess as to why such an inferior algorithm was chosen is because it was convenient. Whatever technique was used to check the age of the images probably lends itself to the current algorithm instead of the one proposed. Either way, it cannot possibly be non-trivial to implement the proposed algorithm in place of the current one. An example routine would be to check the current free-space every 10 minutes, and delete the oldest recorded images/video folder if the free-space percentage was below 5%.

weedy, 2016-09-02:
> An example routine would be to check the current free-space every 10 minutes
You could cron that if this problem is currently killing you

Oberlandix, 2016-09-05:
@John12341234  --> Your suggest, to delete "the oldest files -depending on the free storage" sounds really very nice! I agree to you. I am not an expert in Linux (Crone-Jobs..). So I would love to have this algo as one more simple selection option in the webinterface of motioneyeos. :-)

John12341234, 2016-09-07:
I created a cron job to do it.

`#!/bin/sh
# Delete the oldest folder in /data/output when the sdcard is >= 90% full
# Setup Steps:
# (Copy this script)
# (ssh into motionEyeOS)
# mount -o remount,rw /
# vi /usr/sbin/delete_oldest_dir.sh
#     :set paste
#     i
#     Shift+Ins
# chmod 777 /usr/sbin/delete_oldest_dir.sh
# EDITOR=nano crontab -e
#     @hourly /usr/sbin/delete_oldest_dir.sh
#     Ctrl+O
#     Enter
#     Ctrl+X
ALERT=90
function main_prog() {
while read output;
do
  usep=$(echo $output | awk '{ print $1}' | cut -d'%' -f1)
  if [ $usep -ge $ALERT ] ; then
	 rm -rf "$(find /data/output -maxdepth 1 -type d | sort -r | tail -n 2 | sort | tail -n 1)"
  fi
done
}
# get free space percentage
df -h | grep -vE "^Filesystem|tmpfs|/tmp|/boot|/home|/dev/root" | awk '{print $5}' | main_prog`

Oberlandix, 2016-09-08:
@John12341234  --> sounds really good! Looking forward to test it (when the weather isn´t that nice as now) ;) 

Thanks a lot! :-)

@ccrisan  -> Maybe this would be a good webinterface feature, furhermore ;-)

scoobyd00, 2016-10-03:
This would be a welcomed addition to the webinterface if its possible to add?.

I find that some nights when its 'foggy' outside or the odd spider has built a web across my camera, im getting + 14,000 images!!! which eats away at my storge. I try and keep up with it and delete manually but this addition would be really useful.

John12341234, 2016-10-03:
I have it set up as a cron job and does what it is supposed to now. I think  ccrisan doesn't actually use this software on a regular basis because this is an extremely obvious must-have.

John12341234, 2016-12-11:
The previous script didn't work. Likely because of some old or limited version of something used by motioneye os. Here is the updated script that I've been using for a while now: 

#!/bin/sh
# set alert level 90% is default
ALERT=90
function main_prog() {
while read output;
do
#echo $output
  usep=$(echo $output | awk '{ print $1}' | cut -d'%' -f1)
  if [ $usep -ge $ALERT ] ; then
     echo "Running out of space \"$partition ($usep%)\"" | \
         rm -rf "$(find /data/output -maxdepth 1 -type d | sort -r | tail -n 2 | sort | tail -n 1)"
  fi
done
}
df -h | grep -vE "^Filesystem|tmpfs|/tmp|/boot|/home|/dev/root" | awk '{print $5}' | main_prog

YoungxHelsinki, 2017-03-29:
@John12341234 : Is that script fully functioning? 

I have a SD card that is 8GB and I'd get 10~16GB of pictures a day. 
I would love this feature to be added later on.

Billium28, 2017-04-10:
Yes I would like to know how that script is working. I want to put up 3-4 pi's and let them run. I don't use motion detection and I am more worried about hurting my wifi speeds. Though that may be a silly worry. What would be perfect is an auto restart on the card as it fills.

bortek, 2017-06-01:
Closing his issue due to inactivity. If stil interested please post your question in the forum page.

closed 2017-06-01 by bortek

andreluisos, 2018-08-28:
@SuperShinyEyes 

The script works. However, the cron script isn't making the bash script run, for some reason.

ApvThuanBx, 2019-08-06:
an other option, if you want to delete file only. you should use this code
`
#!/bin/sh
# Delete the oldest folder in /data/output when the sdcard is >= 90% full
# Delete when the sdcard is > 70% full
# Setup Steps:
# (Copy this script)
# (ssh into motionEyeOS)
# mount -o remount,rw /
# vi /usr/sbin/delete_oldest_dir.sh
#     :set paste
#     i
#     Shift+Ins
# chmod 777 /usr/sbin/delete_oldest_dir.sh
# EDITOR=nano crontab -e
#     0 * * * * * /usr/sbin/delete_oldest_dir.sh
#     Ctrl+O
#     Enter
#     Ctrl+X
# set alert level 90% is default
# set end alert level 70% is default
STARTDELETELEVEL=90
ENDDELETELEVEL=70

use=$(df -h | head -n 2 | tail -n 1 | awk '{print $5}' | cut -d '%' -f 1)

flag=0
if [ $use -ge $STARTDELETELEVEL ] ; then
    flag=1
fi

while [ $flag ]; do
    use=$(df -h | head -n 2 | tail -n 1 | awk '{print $5}' | cut -d '%' -f 1)
    if [ $use -ge $ENDDELETELEVEL ] ; then
        find /etc/motioneye/media/ -type f -printf '%T+ %p\n' | sort | head -n 100 | cut -d " " -f 2 | xargs rm
        echo 'Delete'
    else
        echo 'Done'
        flag=0
        break
    fi
done
`

bigjohnson, 2020-02-24:
Here is a script that delete oldest day recorded for all camera.
It checks, and delete only the oldest day directory, if one camera don't has this day, the data are not deleted.
It don't check the space status, you must add the code, or call it from another script
Actually it only display info, and the rm -rf command is commented out for security reasons.

`#!/bin/bash
MOTIONDATA=/var/lib/motioneye
DAYFOUND=false
for CAMERA in $(ls /var/lib/motioneye)
do
        echo ${CAMERA}
        DATA=$(ls "${MOTIONDATA}/${CAMERA}" | sort -r | tail -n 1)
        echo ${DATA}
        DATAS="${DATA}"$'\n'"${DATAS}"
        echo ${DATAS}
        DAYFOUND=true
done
if [ "${DAYFOUND}" == "true" ]
then
        echo "${DATAS}" | sort | head -2 | tail -1
        DELETEDIR=$( echo "${DATAS}" | sort | head -2 | tail -1 )
        echo "${DELETEDIR}"
        for CAMERA in $(ls /var/lib/motioneye)
        do
                if [ -d "${MOTIONDATA}/${CAMERA}/${DELETEDIR}" ]
                then
                        echo "${MOTIONDATA}/${CAMERA}/${DELETEDIR}"
                        ls "${MOTIONDATA}/${CAMERA}/${DELETEDIR}"
#                       rm -rf "${MOTIONDATA}/${CAMERA}/${DELETEDIR}"
                fi
        done
fi
`

starbasessd, 2020-02-24:
Your script only works if you don't make any changes to default settings.
If you change (for example) File Storage path, this script no longer works.
It is a good start, however.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

DaniW42, 2020-06-20:
I had to replace Line 14

`$(ls /var/lib/motioneye)` with `$(ls ${MOTIONDATA})`

for it to work. Thank you @bigjohnson for your work.

bigjohnson, 2020-06-20:
Thanks.
Corrected.

rszabo, 2021-07-10:
An alternate version which I wrote and use on Raspbian. As in the previous examples, I don't extract MOTIONDATA from the software because I don't intend to change it. The script runs every minute using cron. 

Each minute the disk usage is checked and if it's greater than PCENTMAX then the 5 oldest files are removed and empty directories removed. 

`$ vi /home/pi/code/autoclean-motioneye.sh`
```
#!/bin/bash
MOTIONDATA=/var/lib/motioneye
PCENTMAX=85
PCENT="`df -h $MOTIONDATA --output=pcent|tail -n1|sed 's/%//'`" 

if [ "$PCENT" -gt "$PCENTMAX" ]; then
  find $MOTIONDATA -type f -printf '%T+ %p\n' | sort | head -n 5|cut -f2  -d' '|xargs rm -Rf
  find $MOTIONDATA -type d -empty -delete
fi
```
Cron:
`$ sudo crontab -e`

```
*/1 * * * * bash /home/pi/code/autoclean-motioneye.sh
```