motioneye-project/motioneyeos #474: Only detects 2 of 4 devices on capture card

jpenyc, 2016-08-23:
had set up motion first and have all 4 cameras working, then found motioned front end.
installed it and can only add the first two of 4 devices (cameras)
Shows 2 BT878 choices in dropdown on add camera page.

running updated Ubuntu with generic 4 port capture card

ccrisan, 2016-08-23:
I have no idea what "motioned" is. You probably are on the wrong page :)

jpenyc, 2016-08-24:
I wasn't paying attention to the spellcheck that morning after up all night.
I have a Generic 4 channel capture card in each of 2 different Pcs.
I initially installed motion on one and got it running viewing all 4 channels. Then I discovered motioneye and installed that. 
In setting up to add cameras I am only presented with two bt878 choices in the dropdown under the Camera Type Local Camera. 
When selected they display the first two channels on the card.
I installed motioneye in the same manner after setting up motion on the second PC and it shows the same thing, only 2 of the 4 channels show as local.

I hope I explained it better.
Sorry for the confusion.

ccrisan, 2016-08-25:
Normally motionEye looks for /dev/video0, /dev/video1, ... and so on. What does the `ls -l /dev/video*` command say?

jpenyc, 2016-08-25:
`$ ls /dev/video*
/dev/video0`

There are 4 inputs on video0, 0-3. 
In the motion.conf I have to specify each channel.
could it be motioneye is only looking at devices, not channels (inputs)?

ccrisan, 2016-08-26:
Hmm indeed, motionEye doesn't know about channels. Sorry about that.

closed 2017-06-01 by bortek