motioneye-project/motioneyeos #477: No password need... :-O

Oberlandix, 2016-08-26:
Dear motioneyeos friends,

I did install the latest version and set the passwords for admin and the normal user.

Now the surprise: 
When I go to the ip of my raspi with motioneyeos at my HOME NETWORK --> I must log in, correctly.
When I am somwhere else, f.e. at work (at any way NOT at my Home Network!) ---> There is NO Login required!!!! And I have no menue or the possibility to switch user (admin<-->user).

It doesen´t depend on the way i join... If I use the external IP Adress or the "Name-Domain" which I get by dyndns..

Any ideas? It would be very nice!

Thank you,
Oberlandix

joshlukas, 2016-08-27:
How do you access to your cam over internet? There are two possibilities. Via the admin interface or directly to your stream for which you must setup an authentification also. So, do you have port forwarded your admin interface or the stream? The video stream is not password protected by default.

Oberlandix, 2016-08-29:
@joshlukas :+1: Thank you very much for your reply and the information.

I think, (I fear) I have a directly forward to my stream.... Is it real, that EVERYBODY, f.e. a "bad" person who knows my external IP-Adress (and my Cam is configurated with port 8080 or 8081) can see my stream without login..?! :-O

I don´t know how to protect my "native stream"... 

Finally I wan´t to have:
- a protected stream
- a forwarding "just" to the login-site (where i can switch between admin or normal user).

When I am at home, I ll search for these settings again...

THANK you,
Oberlandix

ccrisan, 2016-08-29:
Just forward to port 80 and you'll be fine. The mjpeg streaming is not used by the motionEye user interface.

Oberlandix, 2016-08-30:
@ccrisan  - Thank you very much. Now it works - I see the user interface (although I have now quarrels with the dyn-service. But that´s no problem of motionEye).

I´ll continue playing and trying :-)

closed 2016-08-30 by ccrisan