motioneye-project/motioneyeos #482: Upload of images to FTP (instead of Dropbox or Googledrive): Possible?

Oberlandix, 2016-08-30:
Dear motioneyeos-friends,
another question from a newbie:

It is possible to upload the picture-files (I take a picture every 5 minutes in full resolution) to the two clouds, especially "Google-Drive" and "Dropbox". That is nice -BUT wouldn´t it be nicer to have the possibility to upload the files directly to an ftp-server?

I ask this because that would have -in my view- some advantages:
- no cloude provider and -account is need
- I am the lord of MY files 
- I can give access to some people to my ftp (I am the admin)
- Mostly the webspace is very very large (or even endless big) not just the limitation of gdrive or dropbox
- I can directly connect to the files on my ftp with a own website
- ....

So... Is it really not possible?  Or I've just overlooked that setting/feature? 

Thank you so much,
Oberlandix

ccrisan, 2016-08-30:
Upload to FTP not yet available, but planned.

Oberlandix, 2016-09-01:
@ccrisan  - Perfect! Looking forward! Thank you very much! :-)
Is it also possible in next generation to "group" more photos for a timelapse then just from one day? That would be nice for lager (longer!) projects f.e. building up a house.

Sorry - I am new with motioneyeos (with a lots of questions at the beginn), but I am yet a fan :-) 
Thank you very much!
Oberlandix

moebius911, 2016-09-11:
very good news
I expect to update forward .
Currently the only way to get around the problem is to mount a network drive and to execute a cron job every X minutes to transfer

closed 2017-06-01 by bortek