motioneye-project/motioneyeos #483: Areas in motion detection

xxllo, 2016-08-30:
translated with Google
Is it possible in the next update to divide the motion detection to certain areas ? for example a certain grid in which there is the movement and it only monitors this area ? I have some trees that serves in my picture for monitoring and therefore at each wind impact a motion . would be great if you could divide the picture plane parts to nine to zwöf

joshlukas, 2016-08-31:
Hi. This seems to be possible, however I do not have experience with that. Look at https://github.com/ccrisan/motioneyeos/issues/89 and http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigOptionMaskFile.

ccrisan, 2016-08-31:
It's in the works.

jimuhall, 2016-09-03:
This would be great - constantly getting motion detection trigger from a tree in one quadrant.  Would love to test the beta.

xxllo, 2016-09-07:
Translated with Google
Then only the field of motion detection recording or the whole picture ? The whole picture would be better because I monitor my garden and then all would see .

sapcons, 2016-09-15:
This is in my opinion the no. 1 missing feature of this program. Especially during this time of the year when it is getting windy you receive a lot of alerts from moving trees etc. You can reduce the sensitivity, but this may result in a situation that a real situation is not detected. 
Don't get me wrong: MotionEye is from my perspective a great program. This feature would just make the difference between 'great' and 'leading edge program'.

ccrisan, 2016-09-16:
I am working these days on this feature and it will hopefully be ready within 1-2 weeks.

xxllo, 2016-09-16:
Motioneyeos is the best program in the surveillance sector. Search and experimenting a long time with such programs and can only recommend motioneye

tremichl, 2016-09-18:
I would also appreciate this feature since area_detect in Extra Motion Options doesn't work now for me. Overall great job ccrisan! Thank you very much!

ccrisan, 2016-10-04:
Added in https://github.com/ccrisan/motioneyeos/releases/tag/20160919.

closed 2016-10-04 by ccrisan