motioneye-project/motioneyeos #484: second camera movies won't upload to gdrive after adding adding second camera to rpi3

PresidentNick, 2016-08-30:
I added a second camera to my rpi3 that already has gdrive upload. I chose the same upload settings and got a new authorization key for the second camera. I noticed nothing has been uploaded to that folder on gdrive for the second camera, the first camera is uploading fine.

Here are the log files:

[dmesg.txt](https://github.com/ccrisan/motioneyeos/files/445908/dmesg.txt)
[boot.txt](https://github.com/ccrisan/motioneyeos/files/445910/boot.txt)
[messages.txt](https://github.com/ccrisan/motioneyeos/files/445907/messages.txt)
[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/445911/motioneye.txt)
[motion.txt](https://github.com/ccrisan/motioneyeos/files/445909/motion.txt)

ccrisan, 2016-08-31:
Please try the latest prerelease. There are a few bugs regarding Google Drive that have been fixed recently.

PresidentNick, 2016-09-21:
I've updated the pi to the latest stable 20160915 I still don't see the uploads. Does 20160915 contain the pre-release stuff you mentioned? I didn't update to pre-release at the time since I wasn't sure how to.

ccrisan, 2016-09-21:
20160915 contains all the mentioned fixes. Please try disabling, reenabling and reconfiguring media file upload for both of your cameras. If it still doesn't work, please attach your motioneye.log file again and your config files (after removing sensitive data from them).

PresidentNick, 2016-09-22:
Ok I've disabled, re-enabled for both cameras, I've also rebooted the pi, still don't get the videos on my google drive for the second camera. What is the name of the config file and where is it located?

ccrisan, 2016-09-22:
You can find them at /data/etc/thread-*.conf. Otherwise, from the UI, you can download them as a configuration backup archive.

PresidentNick, 2016-09-22:
Here are the 2 thread-*.conf in that directory.

[thread-2.txt](https://github.com/ccrisan/motioneyeos/files/487419/thread-2.txt)
[thread-1.txt](https://github.com/ccrisan/motioneyeos/files/487420/thread-1.txt)

ccrisan, 2016-09-23:
@PresidentNick could you please attach your motioneye.log now, after you have upgraded to 20160915? You config files look fine.

PresidentNick, 2016-09-23:
[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/489739/motioneye.txt)

ccrisan, 2016-09-24:
@PresidentNick for the second camera, you have enabled Movies and set Recording Mode to Motion Triggered, but the Motion Detection mechanism is disabled. That's the reason why you see no movies for the second camera.

closed 2016-09-24 by ccrisan

PresidentNick, 2016-09-24:
Wow, I'm so dumb, I didn't catch that. Thank you for pointing that out!