motioneye-project/motioneyeos #485: HDMI video output in Fast Network Camera mode

doughadfield, 2016-08-31:
Please can HDMI video output be enabled in Fast Network Camera mode? In this mode, there's no motion capture, so the preview video generated by the GPU can be output to HDMI. This enables a camera to provide both a local high definition picture, and a network stream at lower resolution. HDMI output resolution and frame rate is set via the "video device" section of the GUI, whereas the network streaming properties (resolution, frame rate) should be set via the "Video Streaming" section.

doughadfield, 2016-09-22:
Hi, I've worked on this a bit and have come up with the attached modified raspimjpeg.py program, which now turns on local preview as part of the "init_camera()" function - just a one-line "camera.start_preview()" call at the end of the function. I've fixed the camera resolution at 1280,1024 and used the options.width and options,height values for the network stream ("resize" argument to the "camera.capture_sequence() function". I could have left the resolution values blank for the "PiCamera()" call, which would pick up the native resolution of the screen connected to the HDMI port, but I wasn't sure in my environment how that would behave.
It would be wonderful to be able to set the native resolution (for the arguments to PiCamera()) and separately, the network stream resolution (arguments to "resize" in capture_sequence()). Perhaps you could add a field to the "Video Streaming" section of the config web pane to define the latter, using the "Video Resolution" field of the "Video Device" section to set the native camera resolution (pretty please ;-)). The drop-down list of this field should include "none" which would cause PiCamera() to pick up the native HDMI device resolution.

Another mod was to simplify the generator function "streams_iter()" to be as fast as possible. This has improved CPU consumption on a Zero or ModA by 15% (for frame rate 30fps). 

I hope these initial mod suggestions are helpful to you, but are only suggestions! :-) - I can't begin to understand the entire ecosystem of motioneyeos and whether my mods have negative implications in other areas. But they do seem to work for me :-).

Thanks again for all your hard work in developing this excellent product!!

doughadfield, 2016-09-22:
One further comment on the above mod - I can't find a way to separately control the frame rate of the local HDMI output (native camera settings) and the network stream. In my modified raspimjpeg.py script I've allowed the frame rate to default to 30fps, for minimum latency in the local display. However this puts a great strain on the network link, even when resolution and jpeg quality are dialled right down. Is there a way to separately control frame rate just for the network stream? I spent a lot of time on the "capture_sequence()" call to see if it could be done, but in the end I gave up (looks like it can't be separately controlled, other than something clever in the generator function, which would add significant CPU load). The place I thought it could go would be the "streameye" c program, which manages the network stream to various connected clients. Perhaps this program could take an argument for proportion of frames to drop, and drop 5 in 6, or 2 in 3 frames, to reduce the network frame rate in proportion to the native camera rate. What do you think?
cheers
Doug

ccrisan, 2016-09-22:
I agree with everything you said here. It all makes sense and I'd be happy if you offered to contribute yourself to the sourcecode of raspimjpeg.py yourself. How about I make you a collaborator to streamEye so that you can push the changes yourself?

But whatever you do, please make it all optional. That is, outputting to HDMI is a great idea, but it should happen via a command line switch.

I will take care myself of the C part (streameye.c) to add an optional frame dropping factor. I will also make sure to add an option to motionEyeOS's UI to allow automatically choosing a resolution, in which case, raspimjpeg.py will not be supplied width and height arguments, and will be responsible of choosing its preferred resolution.

doughadfield, 2016-09-22:
Perfect, thanks a lot! - yes, I'll modify my mod to be enabled via command-line switch. I'm not a python programmer (I was a C engineer a very long time ago ;-) so will need a bit of a learning curve to do it properly, but I'll have a go. However, if you're able to add the second resolution selector to the "Video Streaming" web section, then the new functionality of the raspimjpeg.py script would need to be permanent. I'll therefore add a second new command-line switch to add streaming resolution, in anticipation :-)
Cheers

ccrisan, 2016-10-03:
Fixed with ffbc49a5d6d3c57814959217125d7b318fed8ee5.

closed 2016-10-03 by ccrisan

Shaka-60hp, 2017-11-21:
Hi @ccrisan and @doughadfield 
Motioneyeos is great!! And the mod doughadfield did is just the one I need.
I'm new to raspberry world, with some , very little coding experience.

I was wondering what happened with this mod and if I can use it on current version (how to, if so) since it is exactly what I need for a project 

Thanks in advance 
Sorry about my English, I'm from Argentina

doughadfield, 2017-11-22:
Hi, you're right, Motioneyeos IS great!! ;-) - actually, Calin has built this feature into the standard product. Download the latest Motioneyeos for Raspberry Pi (or any version from the last few months) and in the administration settings for the camera, under "general settings" enable the "advanced settings" switch. Then under "Expert Settings" set the "Fast Network Camera" switch. you'll need to reboot after these settings. Then once rebooted, go into the "Video Device" settings and you'll see a switch for "HDMI Preview". Set this switch and you'll get the camera picture on the HDMI port of the Pi. Again, you'll need to re-boot the Pi with an HDMI monitor connected, in order to see the picture (unless you edit the /boot/config.txt file to force HDMI output.
Have fun! :-)
Doug

Shaka-60hp, 2017-11-22:
Hi @doughadfield ,
Thanks for your help, its so kind of you, and you are helping a lot on a very low budget project that is very usefull here 
I was wondering if this is also possible with a webcam instead of the CSI cam, and if the file could be saved , as I read, it is not possible using fast network camera

My project is for a lab that has to manipulate dangerous products and should allow someone to watch on a screen what he is doing in a closed enviroment where he only use tools inside but cannot actually see inside, he only access trough using gloves on the side of a box. Also other people should be able to watch this from a remote computer to take notes and make suggestions. And all of this should be recorded.
 thanks in advance

doughadfield, 2017-11-22:
Hi again, as your usage model sounds very important, I would recommend that you put the very least amount of strain on the Pi running the camera. What I do with my cameras is to run each actual camera Pi as a "fast network camera" with HDMI output. This puts the least amount of CPU strain on the Pi whilst giving the best possible image resolution on the HDMI output and the network stream. I then have another instance of motioneye (not motioneyeos) running on a standard i86 Linux server, which picks up the network streams from the cameras and does all the motion detecting and video recording. If you do this, you will have the most reliable camera picture (which it seems like is very important to you) plus the ability to record the video (on a larger server, with much larger disk storage capability). Also, you should definitely use the Raspberry Pi camera module (or possibly Chinese equivalents - I'm using several of them and they seem fine) if cost is a real issue. This will give you the best picture quality and best reliability, with lowest Pi CPU usage and therefore best picture and streaming quality.
Regards
Doug

Shaka-60hp, 2017-11-22:
Hi @doughadfield , Thanks again, you are helping me a lot
That setup you discribed seems perfect for me, as I can actually get another Rasp (or a netbook , or an old cpu) to do the server task
I'll get the  camera Pi , will do some testing  and then come back to you so you can actually see what you are helping with

Thank you very much, your help is very usefull
Regards
Pablo

ovz93br43v7, 2021-01-08:
Hi @doughadfield @ccrisan Thanks for adding HDMI support can you give me a hint where I can define the HDMI resolution. My display only accept 800x480, I already tried so switch the MJPEG Video Resolution but this doesn't work. 

Countercheck with my laptop and the display was ok....

starbasessd, 2021-01-08:
Add the following lines in /boot/config.txt before inserting in the PI
overscan_right=48
hdmi_group=2
hdmi_mode=14
DO NOT use Notepad to edit. Use Notepad++ or other linux-friendly editor.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2021-01-08:
I used gedit but don't work, I also tried to disable the act led with config.txt same here act led is still working. 
I had in the past also trouble with wpa_supplicant.conf in the boot folder, only way it worked for me was to put the file directly to /etc/wpa_supplicant.conf. Do you know the final destination of the hdmi setting? Somewhere in /etc/?

starbasessd, 2021-01-08:
No, /boot/config.txt can only be edited when the system isn't running, when
it's running most of the system is Read Only. You can control the LED in
the WebGUI when it's running in Expert Settings.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2021-01-08:
I modified my existing sd card and wrote /boot/config.txt on my laptop with following content:
```
# Disable the ACT LED.
dtparam=act_led_trigger=none
dtparam=act_led_activelow=on
# Display HDMI
overscan_right=48
hdmi_group=2
hdmi_mode=14
```
and put it afterwards in the pi. 
For the case that this only work with a fresh installation, I also tried a different sd card, flashed motioneyeos and put /boot/config.txt with the above content on the card. But also don't work. LED on and HDMI wrong resolution.

starbasessd, 2021-01-08:
What are you using to create/edit your SDCard? Linux, Windows, or Mac?
The LED info won't work there.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2021-01-08:
elementaryOS (ubuntu), app is called image burner
Ok o.O LED info is from the wiki --> https://github.com/ccrisan/motioneyeos/wiki/Turn-Off-LEDs