motioneye-project/motioneyeos #486: Two IP when static ip configured

seb49, 2016-09-01:
Hello,

I have an issue after configuring my system with static IP
 this is my /data/etc/static_ip.conf after the modification :

```
static_ip="192.168.2.10/24"
static_gw="192.168.2.247"
static_dns="212.27.40.240"
```

The issue is I have anoter raspberry pi with static IP (192.168.2.8) for my owncloud server. Now I can't access to it. It seem because motioneyeos have 2 IP. Look the ip addr command : 

```
[root@meye-2b71b064 etc]# ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet **127.0.0.1/8** scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether b8:27:eb:71:b0:64 brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.10/24 brd 192.168.2.255 scope global eth0
       valid_lft forever preferred_lft forever
[root@meye-2b71b064 etc]#
```

How can I have only one IP address or how can I setting up the 192.168.2.8 address ?

Regards

seb49, 2016-09-01:
Shame on me, ip addr return 127.0.0.1/ and not 192.168.2.8, the issue is elsewhere

ccrisan, 2016-09-01:
The 127.0.0.1 is a local ip address. It's always like that and that's the way it should be.

closed 2016-09-01 by ccrisan