motioneye-project/motioneyeos #487: Camera unknown message when adding IP cam via RTSP

ruthenensis, 2016-09-01:
Hi all,

I bought 3 IP Cam Foscam FI9800P and tried to install them using a RPi 2 & Motion Eye OS and its "Add Camera" pop-up.
First I made sure that my Foscam cam was RTSP compatible on correct port. 
In Motion Eye OS, when I filled in the correct link "RTSP://IP:554" and correct Username & Password, i've got "Unknown camera" message in red. Due to this message it's impossible for me to validate and add my cam.
I really want to use Motion Eye OS with my cams and as for now it seems impossible....
Please help !!!

ruthenensis, 2016-09-01:
Correction of the previous message : I can now connect to the camera as the "format" was detected but i've got a grey screen saying "unable to open video device"....
As I understand well, because I'm using the OS, the MrDave version could be a solution but at the same time this version is already installed with the OS version. Right ?
So.... Still don't have a solution....

ccrisan, 2016-09-01:
Hi,

1) "Unknown camera" is shown by motionEye when the camera requires Digest authentication (rather than Basic authentication), which cannot be handled (read: tested) by motionEye. So there shouldn't be any problem with that message.

2) The link rtsp://ip:554 is rarely the correct RTSP URL. They usually have a trailing path (mine is /11).

3) Yes, motionEyeOS uses MrDave's fork by default.

ruthenensis, 2016-09-02:
Hi Calin,

First, thanks very much to you for your support which is very appreciated. You're on every topic or issue. Not only mine ! :+1: 
Incredible !!
I have to apologise as I didn't dig deep enough before asking the forum and bothering you....
I found the solution !
the IP & port (554 is the one by default into FOSCAM cam interface) were OK.
But the link must be completed by /videoMain after IP & port.
In addition to that, it seems that TCP protocol is not the good one. I used UCP instead.
Now it's good !! It works fine !!!

Even if my poor RPi 2 is getting hotter & hotter with 3 cam connected !
Does the streaming of these cam could be done in H264 instead of MPEG ? And any solution to see the bandwidth used by them ?
Thanks again !!

ccrisan, 2016-09-02:
The streaming will remain mjpeg as long as the motion daemon will continue to speak mjpeg. The entire solution is based on a stream of individual jpegs so i strongly believe it's going to stay this way.

ruthenensis, 2016-09-02:
OK Thanks again Calin.

Topic closed for me !

closed 2016-09-02 by ruthenensis

bob5731, 2019-09-02:
I now have my camera imported but can't see any video what did I do wrong?