motioneye-project/motioneyeos #490: Trying to connect remote camera

NizzeHulth, 2016-09-04:
What am I doing wrong?

Watch this video

https://1drv.ms/v/s!AgA9Rx7Ag4gGq_J4RPdqfNuLC7dK_g

Best regards
Björn

ccrisan, 2016-09-04:
Make sure the both units run the same version.

NizzeHulth, 2016-09-05:
Okey

I did that and here is the resault!

Link <https://1drv.ms/v/s!AgA9Rx7Ag4gGq_MFlvqreoP3Qt2-hw>

Best regards

Björn

ccrisan, 2016-09-06:
Have you tried clicking outside the "OK" button and see if the Add Camera dialog is happy with the credentials as they are? Also, do you have any special characters in your admin password?

NizzeHulth, 2016-09-06:
Hi

Thanks for your interest!

Yes I have clicked outside the button.

No, the password is digits.

Best regards

Björn

link <https://1drv.ms/f/s!AgA9Rx7Ag4gGq_J3nb1ztLNRvRwumA>  including log
files

closed 2017-06-01 by bortek