motioneye-project/motioneyeos #491: 'protocol rtsp not supported' with motion 3.4.1~rc01 and meye 0.33.2

Microcentillion, 2016-09-04:
Attempting to add an RTSP network camera in MotionEye throws an error 'protocol rtsp not supported'. 

I installed Motion from the packages at https://github.com/Motion-Project/motion/releases - specifically xenialrc01_amd64. Is there any additional configuration needed to support RTSP with MotionEye, or any workaround for adding one?

Thanks in advance!
[image]

Microcentillion, 2016-09-04:
Sorry, I just realized I posted this to meyeos instead of the meye repo.

closed 2016-09-04 by Microcentillion

ccrisan, 2016-09-05:
No worries. The bug has been recently fixed upstream motionEye. If you feel like testing it you can try using the latest commit.