motioneye-project/motioneyeos #497: GDrive missing authorization key

patrickhener, 2016-09-08:
When entering an obtained authorization key and aplying the settings the key vanishes and logging ist stating "missing authorization key". I cannot get the key to remain in the key field.

ccrisan, 2016-09-08:
What version?

patrickhener, 2016-09-08:
Newest one from August with Raspberry Pi 3, Pi NoIR v2, Static IP.

Testing the key states that everything is alright with the key: success. But configuration apply will not keep the key.

Same thing for Dropbox inbetween.

ccrisan, 2016-09-08:
Applying the settings shouldn't include the authorization key, if you have tested the connectivity. The authorization key is used only once with the first request to google drive or dropbox. Therefore it is normal that it disappears.

patrickhener, 2016-09-08:
Okay the why is the log stating "Missing authorization key" everytime motion is detected and therefor pictures and videos should be uploaded to GoogleDrive?
The pictures and videos are not uploaded though.

ccrisan, 2016-09-08:
Somehow the access token must have been lost as well. A full debug log would help but I don't suppose you have enabled debugging. At this moment I recommend repeating the authorization process to Google Drive starting with obtaining a new authorization key.

patrickhener, 2016-09-08:
Did that like a hundret times. Let me see if I can fetch the log real quick.

ccrisan, 2016-09-08:
Please enable debugging in expert settings than then repeat the procedure.

patrickhener, 2016-09-08:
`1970-01-01 01:01:11: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 6.50ms
1970-01-01 01:01:14: [motioneye]    DEBUG: loading upload services state from "/data/etc/uploadservices.json"...
1970-01-01 01:01:14: [motioneye]    DEBUG: loaded upload service "dropbox" for camera with id "1"
1970-01-01 01:01:14: [motioneye]    DEBUG: loaded upload service "gdrive" for camera with id "1"
1970-01-01 01:01:14: [motioneye]    DEBUG: redirected to authorization url https://accounts.google.com/o/oauth2/auth?scope=https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fdrive&redirect_uri=urn%3Aietf%3Awg%3Aoauth%3A2.0%3Aoob&response_type=code&client_id=349038943026-m16svdadjrqc0c449u4qv71v1m1niu5o.apps.googleusercontent.com&access_type=offline
1970-01-01 01:01:14: [motioneye]    DEBUG: 302 GET /config/1/authorize/?service=gdrive&_username=admin&_signature=071a6d116f0add07f0ef37ca04e9e1dcc68c98fd (192.168.138.1) 6.44ms
1970-01-01 01:01:16: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 3.16ms
1970-01-01 01:01:18: [motioneye]    DEBUG: running cleanup process...
1970-01-01 01:01:18: [motioneye]    DEBUG: cleaning up pictures...
1970-01-01 01:01:18: [motioneye]    DEBUG: cleaning up movies...
1970-01-01 01:01:18: [motioneye]    DEBUG: cleanup done
1970-01-01 01:01:18: [motioneye]    DEBUG: running garbage collector for mjpg clients...
1970-01-01 01:01:21: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 8.83ms
1970-01-01 01:01:22: [motioneye]    DEBUG: setting config for camera 1...
1970-01-01 01:01:22: [motioneye]    DEBUG: loading upload services state from "/data/etc/uploadservices.json"...
1970-01-01 01:01:22: [motioneye]    DEBUG: loaded upload service "dropbox" for camera with id "1"
1970-01-01 01:01:22: [motioneye]    DEBUG: loaded upload service "gdrive" for camera with id "1"
1970-01-01 01:01:22: [motioneye]    DEBUG: saving upload services state to "/data/etc/uploadservices.json"...
1970-01-01 01:01:22: [motioneye]    DEBUG: adding task "invalidate_uploadservices" in 0 seconds
1970-01-01 01:01:22: [motioneye]    DEBUG: saving tasks to "/data/etc/tasks.pickle"...
1970-01-01 01:01:22: [motioneye]    DEBUG: writing wifi settings to /data/etc/wpa_supplicant.conf: enabled=False, ssid=""
1970-01-01 01:01:22: [motioneye]    DEBUG: linking "/data/etc/localtime" to "/usr/share/zoneinfo/Europe/Berlin"
1970-01-01 01:01:22: [motioneye]    DEBUG: writing ip settings to /data/etc/static_ip.conf: type=static, addr=192.168.138.100, mask=255.255.255.0, gw=192.168.138.1, dns=192.168.138.1
1970-01-01 01:01:22: [motioneye]    DEBUG: saving service settings: ftp=True, ftp_auth=True, ftp_writable=False, smb=True, smb_auth=True, smb_writable=False, ssh=True
1970-01-01 01:01:22: [motioneye]    DEBUG: setting FTP mode: auth
1970-01-01 01:01:22: [motioneye]    DEBUG: setting SMB mode: auth
1970-01-01 01:01:22: [motioneye]    DEBUG: setting SSH mode: enabled
1970-01-01 01:01:22: [motioneye]    DEBUG: writing watch settings to /data/etc/watch.conf: watch_link=True, watch_link_timeout=20, watch_connect=False, watch_connect_host=www.google.com, watch_connect_port=80, watch_connect_retries=3, watch_connect_timeout=5, watch_connect_interval=20
1970-01-01 01:01:22: [motioneye]    DEBUG: hostname cam-tuere written to /data/etc/hostname
1970-01-01 01:01:22: [motioneye]    DEBUG: writing date settings to /data/etc/date.conf: method=http, host=google.com, ntp_server=, timeout=10, interval=900
1970-01-01 01:01:22: [motioneye]    DEBUG: writing motioneye settings to /data/etc/motioneye.conf: port=80, base_path=, motion_binary=/usr/bin/motion, motion_keep_alive=False, debug=True
1970-01-01 01:01:22: [motioneye]    DEBUG: writing OS settings to /data/etc/os.conf: prereleases=True
1970-01-01 01:01:23: [motioneye]    DEBUG: reading board settings from /boot/config.txt
1970-01-01 01:01:23: [motioneye]    DEBUG: board settings: gpu_mem=128, camera_led=True
1970-01-01 01:01:23: [motioneye]    DEBUG: reading main config from file /data/etc/motion.conf...
1970-01-01 01:01:23: [motioneye]    DEBUG: writing main config to /data/etc/motion.conf...
1970-01-01 01:01:23: [motioneye]    DEBUG: reading camera config from /data/etc/thread-1.conf...
1970-01-01 01:01:23: [motioneye]    DEBUG: writing camera config to /data/etc/thread-1.conf...
1970-01-01 01:01:23: [motioneye]    DEBUG: motion needs to be restarted
1970-01-01 01:01:23: [motioneye]    DEBUG: stopping motion
1970-01-01 01:01:26: [motioneye]    DEBUG: updating SMB mounts
1970-01-01 01:01:26: [motioneye]    DEBUG: listing smb mounts...
1970-01-01 01:01:26: [motioneye]    DEBUG: starting motion
1970-01-01 01:01:26: [motioneye]    DEBUG: using motion binary "/usr/bin/motion"
1970-01-01 01:01:28: [motioneye]    DEBUG: motion detection disabled by config for camera with id 1
1970-01-01 01:01:28: [motioneye]    DEBUG: disabling motion detection for camera with id 1
1970-01-01 01:01:28: [motioneye]    DEBUG: 200 POST /config/0/set/?_=1473330395532&_username=admin&_signature=1bce0d256608780b1971b0feee058800072d9c29 (192.168.138.1) 5748.96ms
1970-01-01 01:01:28: [motioneye]    DEBUG: recevied relayed event stop for thread id 1 (camera id 1)
1970-01-01 01:01:28: [motioneye]    DEBUG: clearing motion detected for camera with id 1
1970-01-01 01:01:28: [motioneye]    DEBUG: 200 POST /_relay_event/?_username=admin&event=stop&thread_id=1&_signature=40d614321b36048958eda9795238c7a32306faec (127.0.0.1) 6.21ms
1970-01-01 01:01:28: [motioneye]    DEBUG: recevied relayed event stop for thread id 1 (camera id 1)
[... cut ...]
1970-01-01 01:01:28: [motioneye]    DEBUG: getting disk usage for path /data/output/Camera1...
1970-01-01 01:01:28: [motioneye]    DEBUG: 200 GET /config/1/get/?force=true&_=1473330401300&_username=admin&_signature=57ab029004ae08dadb64ac5ea5f6804c17e69ffd (192.168.138.1) 19.80ms
1970-01-01 01:01:28: [motioneye]    DEBUG: creating mjpg client for camera 1
1970-01-01 01:01:28: [motioneye]    DEBUG: mjpg client for camera 1 connected on port 8081
`

That is the complete log of me putting in the key and applying the settings.
Next for the motion and upload part.

patrickhener, 2016-09-08:
`1970-01-01 01:03:28: [motioneye]    DEBUG: 200 POST /_relay_event/?_username=admin&event=picture_save&thread_id=1&_signature=4124cea1e87eb7bab3f16af58555c9f8e819ee70 (127.0.0.1) 22.20ms
1970-01-01 01:03:28: [motioneye]    DEBUG: recevied relayed event picture_save for thread id 1 (camera id 1)
1970-01-01 01:03:28: [motioneye]    DEBUG: adding task "upload_media_file(/data/output/Camera1/1970-01-01/01-03-25.jpg)" in 5 seconds
1970-01-01 01:03:28: [motioneye]    DEBUG: saving tasks to "/data/etc/tasks.pickle"...
1970-01-01 01:03:28: [motioneye]    DEBUG: 200 POST /_relay_event/?_username=admin&event=picture_save&thread_id=1&_signature=4124cea1e87eb7bab3f16af58555c9f8e819ee70 (127.0.0.1) 9.39ms
1970-01-01 01:03:28: [motioneye]    DEBUG: recevied relayed event picture_save for thread id 1 (camera id 1)
1970-01-01 01:03:28: [motioneye]    DEBUG: adding task "upload_media_file(/data/output/Camera1/1970-01-01/01-03-26.jpg)" in 5 seconds
1970-01-01 01:03:28: [motioneye]    DEBUG: saving tasks to "/data/etc/tasks.pickle"...
1970-01-01 01:03:28: [motioneye]    DEBUG: 200 POST /_relay_event/?_username=admin&event=picture_save&thread_id=1&_signature=3ad7321e67cfc18974cd487346dcbc66f8f6b590 (127.0.0.1) 8.66ms
1970-01-01 01:03:28: [motioneye]    DEBUG: recevied relayed event picture_save for thread id 1 (camera id 1)
1970-01-01 01:03:28: [motioneye]    DEBUG: adding task "upload_media_file(/data/output/Camera1/1970-01-01/01-03-26.jpg)" in 5 seconds
1970-01-01 01:03:28: [motioneye]    DEBUG: saving tasks to "/data/etc/tasks.pickle"...
1970-01-01 01:03:28: [motioneye]    DEBUG: 200 POST /_relay_event/?_username=admin&event=picture_save&thread_id=1&_signature=3ad7321e67cfc18974cd487346dcbc66f8f6b590 (127.0.0.1) 6.94ms
1970-01-01 01:03:28: [motioneye]    DEBUG: executing task "upload_media_file(/data/output/Camera1/1970-01-01/01-03-22.jpg)"
1970-01-01 01:03:28: [motioneye]    DEBUG: loading upload services state from "/data/etc/uploadservices.json"...
1970-01-01 01:03:28: [motioneye]    DEBUG: loaded upload service "dropbox" for camera with id "1"
1970-01-01 01:03:28: [motioneye]    DEBUG: loaded upload service "gdrive" for camera with id "1"
1970-01-01 01:03:28: [motioneye]    DEBUG: executing task "upload_media_file(/data/output/Camera1/1970-01-01/01-03-22.jpg)"
1970-01-01 01:03:28: [motioneye]    DEBUG: saving tasks to "/data/etc/tasks.pickle"...
1970-01-01 01:03:28: [motioneye]    DEBUG: gdrive: uploading file "/data/output/Camera1/1970-01-01/01-03-22.jpg" to gdrive
1970-01-01 01:03:28: [motioneye]    DEBUG: gdrive: size of "/data/output/Camera1/1970-01-01/01-03-22.jpg" is 0.077MB
1970-01-01 01:03:28: [motioneye]    DEBUG: running garbage collector for mjpg clients...
1970-01-01 01:03:28: [motioneye]    DEBUG: gdrive: mime type of "/data/output/Camera1/1970-01-01/01-03-22.jpg" is "image/jpeg"
1970-01-01 01:03:28: [motioneye]    DEBUG: gdrive: finding folder id for location "/Ueberwachung/1970-01-01"
1970-01-01 01:03:28: [motioneye]    ERROR: gdrive: missing authorization key
1970-01-01 01:03:28: [motioneye]    ERROR: failed to upload file "/data/output/Camera1/1970-01-01/01-03-22.jpg" with service gdrive: missing authorization key
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 707, in upload_media_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 91, in upload_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 186, in upload_data
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 241, in _get_folder_id
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 253, in _get_folder_id_by_path
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 273, in _get_folder_id_by_name
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 318, in _request
Exception: missing authorization key
1970-01-01 01:03:28: [motioneye]    DEBUG: gdrive: uploading file "/data/output/Camera1/1970-01-01/01-03-22.jpg" to gdrive
1970-01-01 01:03:28: [motioneye]    DEBUG: gdrive: size of "/data/output/Camera1/1970-01-01/01-03-22.jpg" is 0.077MB
1970-01-01 01:03:28: [motioneye]    DEBUG: gdrive: mime type of "/data/output/Camera1/1970-01-01/01-03-22.jpg" is "image/jpeg"
1970-01-01 01:03:28: [motioneye]    DEBUG: gdrive: finding folder id for location "/Ueberwachung/1970-01-01"
1970-01-01 01:03:28: [motioneye]    ERROR: gdrive: missing authorization key
1970-01-01 01:03:28: [motioneye]    ERROR: failed to upload file "/data/output/Camera1/1970-01-01/01-03-22.jpg" with service gdrive: missing authorization key
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 707, in upload_media_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 91, in upload_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 186, in upload_data
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 241, in _get_folder_id
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 253, in _get_folder_id_by_path
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 273, in _get_folder_id_by_name
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 318, in _request
Exception: missing authorization key
`

[... cut ...]
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 186, in upload_data
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 241, in _get_folder_id
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 253, in _get_folder_id_by_path
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 273, in _get_folder_id_by_name
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 318, in _request
Exception: missing authorization key
`

ccrisan, 2016-09-08:
Please attach the file. This is impossible for me to follow. Also your network connection seems to be problematic as the date/time is not updated.

patrickhener, 2016-09-08:
[Applying_Settings.txt](https://github.com/ccrisan/motioneyeos/files/461390/Applying_Settings.txt)
[upload_Error.txt](https://github.com/ccrisan/motioneyeos/files/461391/upload_Error.txt)

Pinging google will work. No clue why my clock is not synchronised. It was a few boots ago.

ccrisan, 2016-09-08:
boot.log will tell you what's wrong with the date, btw.

patrickhener, 2016-09-08:
Setting current date using http: failed

Well thats not really detailed ;) but its okay. I guess it will come back a few boots away

ccrisan, 2016-09-08:
Hmm I will have to look into this. Have you tried testing the service first before applying it? Does it behave the same way?

patrickhener, 2016-09-08:
I tested it a few times and everytime it states that it'll work. But the key vanishes by testing, too. So i repaste it in, apply, and gone.

ccrisan, 2016-09-08:
Try testing and then applying w/o pasting the key again. The test will act as a first request.

patrickhener, 2016-09-08:
Tried that. Still missing authorization key. Date and time came back after switching to a wireless lan with the camera. Check stated success, too.

ccrisan, 2016-09-08:
I will have to run some tests myself. I really hate all this OAuth crap.

patrickhener, 2016-09-08:
Okay thanks. I'm looking forward to your results. Thanks in advance.

stllunatic, 2016-09-08:
I am getting similar results from my setup.(Newest one from August with Raspberry Pi 3, Static IP, Eth0 connected). The testing button says it fails and the auth key disappears. Only difference is the pictures upload and can be seen in the Google Drive for the most part. I have noticed the files aren't loading as often as I have it configured. For me, if I give it a few hours and come back; the files start appearing in the drive.

ccrisan, 2016-09-11:
The upload services are indeed (still) broken. I have managed to reproduce the issue and I know why this happens. However things are a bit more complex and there are parts that need to be rewritten. I'll make sure to take care of this issue asap and will keep you posted.

ccrisan, 2016-09-12:
https://github.com/ccrisan/motioneye/commit/8bfa621874a5078928687d89f3d517aeecfea7f0 fixes many upload services bugs, including this one.

closed 2016-09-12 by ccrisan

patrickhener, 2016-09-14:
Hey There.
How would i get the change into my motioneyeos setup? Update button says I'm on the newest release, which is right.

ccrisan, 2016-09-14:
You can't do that. I'll try to push an update some these days.

patrickhener, 2016-09-14:
Okay thanks a lot =)