motioneye-project/motioneyeos #499: Question mark in Web UI doesn't work

bradyjoh, 2016-09-10:
I'm trying to find out more information about the settings in the web GUI, but clicking the question mark in Chrome or Safari on OS X doesn't seem to do anything. On Android Chrome, the question marks don't show up at all. Does anybody else have this issue?

ccrisan, 2016-09-11:
The question marks are supposed to activate on hover, not on click. I have plans to change this behaviour, but it's not one of the priorities.

closed 2017-06-01 by bortek

surak, 2019-05-07:
It's not just that - in safari, the hover disappears as soon as you reach the question mark. That is, it works when you hover on the circle, as long as the mouse pointer is not actually touching the question mark character.

I use it for a while and I only noticed that hover works at all now. Not to say that hover is useless on non-mouse devices.