motioneye-project/motioneyeos #501: how to reset admin account password once changed?

DrCord, 2016-09-12:
how does one do a "forgot my password" reset on the admin account, I got locked out of it by changing the password in the UI...

ccrisan, 2016-09-12:
There's no such functionality. You'll need to pull out your SD card, plug it in to a Linux machine and edit the motion.conf file to change the password using a text editor.

DrCord, 2016-09-12:
Thanks for the quick answer. To find the path of the the file to edit once you have the microSD card mounted as an external drive in a linux machine you can run `find / -name motion.conf` from the terminal.

Worked great for me to reset this way as I have other linux computers around.

You may want to either consider adding a password reset feature or a second field that has to match when changing the password as it is very easy to lock yourself out with a typo when changing the admin password...and others may not have another linux box around or possibly not have physical access to the box running motioneye anymore as it is mounted in a project with the camera or some such...

Thanks for your hard work on this great piece of software!

closed 2016-09-12 by DrCord

Just-another-pleb, 2018-11-03:
I have the same problem.

After reading all the posts, there is a serious disconnect between the files named and the actual files.
Here is is /etc/motion/motion.conf

In another thread it is:
[https://github.com/ccrisan/motioneyeos/wiki/FAQ#how-can-i-recoverreset-the-admin-password](url)

> Unfortunately there's no easy way to recover or reset your admin password. You'll need a computer running Linux and capable of mounting your ext4 partition from the SD card. Identify the data partition (normally the largest one) and look for the file /etc/motion.conf (on a live system it's /data/etc/motion.conf). Just clear out the value at the end of the line starting with # @admin_password.

Which is also wrong.

The file to be edited - as I have seen so far in my travels - is:
/etc/motioneye/motion.conf

Can someone please get the documentation up to date so everything is saying the correct thing?
It is painful for me (having seemingly forgotten) the ADMIN password to get it reset.

At this time I am still not 100% sure it is going to work, but at least /etc/motioneye/motion.conf has a line which looks like the one mentioned:
> Just clear out the value at the end of the line starting with # @admin_password

sredeck, 2019-04-20:
i had once forget motioneye password so what i doing is edit file in /etc/motioneye/motion.conf, find @admin_password. you may deleted right side @admin_password and change it....voalaaaa.

sredeck, 2019-04-20:
no need to pull out sd card. just find edit file in /etc/motioneye/motion.conf. right side of @admin_password you may change it

Henchman-21, 2020-12-30:
How do you edit the file without removing the SD card, or knowing the admin pass to SSH in?

starbasessd, 2020-12-30:
If you can access the WebGUI, you can reset it in there.
Otherwise you need to access the SDCard.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

nmg196, 2021-01-15:
I'm really stuck with this. I can't find any way to reset the password from a Windows PC. I've even tried Ubuntu but I can't work out how to 'see' or mount the SD card as I have very little Linux experience.  Is there any way I can edit a file on the SD card with Windows?  

I also don't understand the comment above where it says if you can see WebGUI you can reset it there. I can't see a reset option in MotionEye anywhere. Only Login.

starbasessd, 2021-01-15:
In the grand scheme of things, it's faster just to re-image the SDCard.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mwy6283, 2021-01-18:
Hi, is there a solution to resetting login details . just installed HA and Motioneye put the login details and was called away and forgot the details.  I thought i could set up HA again( different details & ip ) but still unable to as it must be linked to your location ??. I really need some help as i new to the system like lInux . if you could point me in the right direct i would be grateful and obviously lean something.

nmg196, 2021-01-18:
Yeah I ended up reimaging the SD card.  It's a shame there's no password reset mechanism.   Also take a backup of the settings so if you ever have to reimage again, you don't have to start from scratch.  I found it took several days to tune the motion settings so it worked reliably without loads of false positives so a lot of time is wasted if you have to reimage the SD card and don't have a backup.

mwy6283, 2021-01-18:
nmg196, 
Thanks for quick response.  Motioneye is installed on my Imac. It was a silly mistake on my part . So reimaging would not help me at this point. I tried reinstalling HA on again with different credential but this did not work. When i loaded Motioneye it requested a username and password not  register. So i am spending some much time trying to find a solution.as i haven't  loaded my cameras yet. If you say it is good  I would love to use. There must be a solution , I just hope that the developer replies with some advice.

starbasessd, 2021-01-18:
You describe several different situations.
For motionEyeOS, the easiest solution is to re-image, unless you can read &
write EXT4 partitions.
For HA, you need to contact the HA Community forums, as FRENCK made a bunch
of changes to motionEye to fit their model, and hasn't responded to me for
requests for info.
For motionEye installed from the CCrisan github install instructions (for
Ubuntu for example), the file you can edit is located at
:/etc/motioneye/motion.conf
place a # in front of @admin_password
reboot
Password should be set back to blank

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pawelnguyen, 2021-02-25:
For HA if you forgot your password and you're using MotionEye in a docker container, you can find it here:
```bash
docker ps # find your motion eye addon container name
docker exec -it addon_aXXXXX_motioneye /bin/bash
cat /data/motioneye/motion.conf # your pass will be in `# @admin_password` line
```

nmg196, 2021-03-01:
How would you manage to get a shell prompt if you don't know the password?

starbasessd, 2021-03-01:
You can't, with motionEyeOS, by itself.
If you can use a Linux machine (iron or VM) and mount mmcblk0p2, you can
edit the file /etc/motioneye.conf or mount mmcblk0p3 and edit
/data/etc/motioneye.conf
put a # in front of the admin line
Other than that, re-image.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-01:
For motionEyeOS, the easiest solution is to re-image, unless you can read &
write EXT4 partitions.
/data/etc/motion.conf is in mmcblk0p3 (sudo mount /dev/mmcblk0p3 /tmp in a ubuntu linux)
place a # in front of @admin_password
Password should be set back to blank
then sudo umount /tmp
eject the sd card, put it back in the PI, and boot.