motioneye-project/motioneyeos #503: Raspberry Zero + GSM

ram-sh, 2016-09-12:
Dear MotionEyeOs,

Let me start by saying that your project is awesome! I highly appreciate your work!

I'm using the MotionEyeOs running on Raspberry Zero, currently it receives its IP via WiFi and I need it to get the data via GSM, specifically using the [Adafruit FONA 808 - Mini Cellular GSM + GPS Breakout](https://www.adafruit.com/product/2542) , I'm quite lost here since this is the first project I'm using Linux as an OS (I'm an Arduino/Netduino developer), there's a nice project [Make a Cellular-Connected GPS with the Fona & Pi Zero](https://github.com/InitialState/fona-pi-zero/wiki) that demonstrate the setup between Zero & Fona, would it be possible to do the same in addition to MotionEyeOs in some way?

Many thanks in advance!
Best,
Ram

bortek, 2017-06-01:
Closing his issue due to inactivity. If stil interested please post your question in the forum page.

closed 2017-06-01 by bortek

MahbbRah, 2020-10-19:
Well I'm interested about this. nobody ever mentions the hard part!