motioneye-project/motioneyeos #505: MotioneyeOS and TP-LINK TL-WN722N

tekgeek1, 2016-09-13:
So I am fairy new to motioneye but think it is some great work!  I am having an issue the the TP-LINK wireless card.  I have things setup to use wireless from the motioneye config screen - have my SSID and PW entered.  So I apply the settings and reboot.  But when it comes up, it has the static IP I gave it in the wireless config BUT it still on eth0.

Suggestions?

Thanks!

ccrisan, 2016-09-13:
The suggestion is to unplug your ethernet cable or otherwise the static IP will always go to it as it's the preferred connection.

closed 2016-09-13 by ccrisan

tekgeek1, 2016-09-13:
Hmmm - tries that still no luck.  I will try a different wireless card
since it looks like from the boot log that the divers are detected.

-Bruce

ccrisan, 2016-09-13:
Please attach your boot log here.

tekgeek1, 2016-09-13:
See attached boot log.  I cleaned the log out - plugged in the TP-Link and
you can see it is in an endless boot loop looking for a connection - then I
unplugged it and plugged the ethernet back in.

Thanks!
-Bruce